rogerclarkmelbourne/OpenGD77 #452: Talk Permit Tone?

G0FHM, 2020-02-28:
Would it be possible to have it back please? (assuming it's an easy enough job to implement!) Using a RT91 amp when mobile. Radio/Amp is out of visual range when driving. Audible confirmation that I have the repeater would be nice.

Thanks

rogerclarkmelbourne, 2020-02-28:
You mean this ??

https://www.opengd77.com/viewtopic.php?f=13&t=686

G0FHM, 2020-02-28:
Er, yes!

Thank you

testbird, 2020-02-28:
> I tried gettings the radio to receive on the alternate timeslot to the one it was transmitting on, but I was getting some strange effects on the DMR audio when I did this, which is a shame as it would have allowed us to monitor whether the repeater was still transmitting.

Oh yes, that would be cool. Figuring out enough of the DMR chip register configuration of the memory written and read might even allow to confirm that it's exactly the own data frames coming back from the repeater. (Talk permit tone if transmission is confirmed.)

And likewise it could also allow, to re-transmit any frames received on a [auto-start SFR mode channel](https://github.com/hamdv/OpenGD77/issues/57#issuecomment-573774154). :-)

rogerclarkmelbourne, 2020-02-29:
Closing as Tx Permit has already been released as an experimental build

closed 2020-02-29 by rogerclarkmelbourne