rogerclarkmelbourne/OpenGD77 #472: Support Baofeng RD-5R

rogerclarkmelbourne, 2020-03-08:
The Baofeng RD-5R seem to use the same chipset as the GD-77 and DM-1801, except for the display.

The radio contains the following IC's
MK22FN512 MCU
HR-C6000 DMR DSP
AT1846S Transceiver
24C512 EEPROM
25Q80  Flash
TDA2822  Audio amplifier
Power supply controller IC

The display has ZYEG1284805-FPC written on the ribbon cable.

I think this refers to the display being 128 x 48 pixels.
The display is connected via a 14 wide ribbon cable, so this display is likely to be sent parallel data by 8 of the wires. Some of the wires will power the backlight, and one or more will be the GND connection

However at the moment, I have not been able to find any data on this specific display

rogerclarkmelbourne, 2020-03-08:
My radio has "RD-5R" on the sticker on the main chassis.

On the PCB it has 

5R-DMR-V2.2  
20180911

I bought it from Radioddity 

https://radioddity.com/products/radioddity-x-baofeng-rd-5r-dmr-ham-amateur-two-way-radio?_pos=6&_sid=369e7279b&_ss=r

f1rmb, 2020-03-08:
Hi Roger,

   Okay, thanks for the info

Cheers.
---
73 de Daniel, F1RMB
http://f1rmb.blog.free.fr

TecnoTalarn, 2020-03-10:
http://www.customlcddisplay.com/uploads/201816158/Datasheet.pdf <--Surely this is not the same display, but I see that the data goes in series and not in parallel. This interface has 12 pins and not 14 as yours.
http://www.customlcddisplay.com/cog-module/128x48-cog-module/128x48-graphic-cog-module.html

rogerclarkmelbourne, 2020-03-11:
@TecnoTalarn 

OK.

I presumed that because the display had a lot of connections, that it was a parallel display, but I agree that I was probably wrong, and this is a serial display.

ew1adg, 2020-03-30:
Hi, recently, i was make patches to get it work on DM-5R (visually the same radio)
And before i will made pull request, i am need to perform some testing and codereview.
If somebody have a precise rf power meter please help me to determine correct **fractional power coeffs**. Thanks 

https://github.com/ew1adg/OpenGD77 branch:dm5r_support

rogerclarkmelbourne, 2020-03-30:
I have a power meter, which is accurate to approximately 50mW steps.

This is enough accuracy for the firmware because each radio has a lot of variation, and the fractional steps are only around 90% accurate across different radios even when the 1W and 5W power points are set using the calibration screen in the CPS

rogerclarkmelbourne, 2020-03-30:
BTW. I think this link to your repo is better

https://github.com/ew1adg/OpenGD77/tree/dm5r_support

rogerclarkmelbourne, 2020-03-30:
Do you think it will work with my RD-5R?

ew1adg, 2020-03-30:
> Do you think it will work with my RD-5R?
I think, it will work.
Note:
 "A/B" acts as a "LEFT" and "band" acts as a "RIGHT"
 "VFO/MR" - switch between VFO and Channel modes 
to switch VFO use long pressed "A/B" button
to enter Quick menu use long pressed "VFO/MR" button

rogerclarkmelbourne, 2020-03-30:
OK. Thankyou.

I will try it tomorrow.

rogerclarkmelbourne, 2020-04-30:
Support has mostly now been added. Just a few bugs need to be fixed

closed 2020-04-30 by rogerclarkmelbourne