rogerclarkmelbourne/OpenGD77 #497: TX power *-

bv3un, 2020-03-22:


bv3un, 2020-03-22:
The transmit power setting in each channel is invalid . it will overwrite by Function + Right(Left)
that means you need to set every time after change channel
for example 
using 500mw at home for MMDVM channel and need to 5W for repeater channel outside

N4NQY, 2020-03-23:
Power seems to default to 1w (at least on my GD-77) and cannot be changed.  Power set in CPS ignored.

rogerclarkmelbourne, 2020-03-23:
This is by design

The Power setting per channel is ignored, because the codeplug data only has 2 levels, whereas the firmware supports 10 levels

closed 2020-03-23 by rogerclarkmelbourne

bv3un, 2020-03-23:
Thanks for answer,so I need to set power every time after change channel Right ?Thanks

傳送自 Yahoo奇摩電子信箱 Android 版

rogerclarkmelbourne, 2020-03-23:
Yes. If you need a different power for each channel.

Its impossible to support 10 power levels with the CPS format. Hence per channel power setting will never be supported while the official codeplug format is used.

The power level dropdown menu will eventually be removed from the OpenGD77 CPS, but at the moment I have higher priorities

bv3un, 2020-03-23:
Got itThanks

傳送自 Yahoo奇摩電子信箱 Android 版

N4NQY, 2020-03-29:
Would it be possible to support at least a High and Low setting. Any other choice could be done by Function Right. 

Mostly the High (whatever that may be) Is a good choice but Not wanted when you go to use the radio with a hotspot at home or in a vehicle. 

I agree that there are others ways to accomplish this however, so it’s not a “must”. 

73

jb  N4NQY

mti-sk, 2020-07-22:
Hi, 
It is possible to set the power for HOTSPOT mode, but only before entering this mode. standard procedure Fn2 button + left or right arrow.
from 50mW-250mW-500mW-750mW-1W-2W-3W-4W-5W

However, the problem is, as I mention in my issue https://github.com/rogerclarkmelbourne/OpenGD77/issues/688 ,
that it is probably not possible to use a power greater than 250mW, because the HOTSPOT mode is terminated spontaneously.

f1rmb, 2020-07-22:
No, how to define the hotspot power setting is described in the user guide. 

Please, don't add posts to closed issues.