rogerclarkmelbourne/OpenGD77 #552: rx ctcss not usable anymore bug

riku2015, 2020-04-22:
after doing something on ctcss(i think it was update where was "improvement on ctcss" the squelch now will close, not even opening the squelch will help.
Using RX ctcss listening our repeater when someone talking the radio will all time close sq. not usable right now, only way to go around this is disable rx ctcss(as we have DMR/FM hybrid repeater this is not an option).
This is easy to reproduce, use 2 radios other put tx ctcss and other radio rx ctcss and YELL to radio, you will see other radio where is rx-ctcss will breake up filtering high modulated sound.

rogerclarkmelbourne, 2020-04-29:
Can't replicate

closed 2020-04-29 by rogerclarkmelbourne

riku2015, 2020-04-30:
there is youtube video about this on forums https://www.opengd77.com/viewtopic.php?p=7907#p7907
also there is also a new fix for this https://www.opengd77.com/viewtopic.php?p=7899#p7899
Now radio use with ctcss will not possible at all.
It is on code the problem not on radio, so you are doing something wrong while replicating it. 
look the code you will understand the problem. As wrote on forums im trying to help on severe problem and all is just ignoring me. AGAIN. sooner or later other users will notice this and after that action is taken?

f1rmb, 2020-04-30:
Seriously ?!
Does anyone here owe you something ?
What don't you understand in "Can't replicate" ?