rogerclarkmelbourne/OpenGD77 #600: Question: what debugger are you using?

redfast00, 2020-05-07:
On the wiki, it says that a JLink Edu is recommended for debugging during firmware development, but since these are rather expensive, I'm looking for a more affordable option. I saw that you have a purple board soldered to your radio, what board is it? Does it work as well as the JLink? Do you recommend it?

rogerclarkmelbourne, 2020-05-07:
After I unlock with the JLink Edu I use a STM32 Blue pill I modified to act as a JLink OB

https://github.com/GCY/JLINK-ARM-OB

or

https://github.com/idrnyu/dllGetSt072JlinkOb

But I can't get the JLink OB to unlock the MK22, because I can't get the reset pin to work.
Its possible I have not wired up the STM32 correctly. It RST may be on PA0 and PB1.

The JLink V8 and JLink V9 clones, the ones in the large black boxes, seem to initially work OK, but the problem I had with my Jlink V8 clone, was that it kept getting bricked, and I had to reflash its internal firmware using another debugger.

I suspect the JLink software detects its a clone and bricks it.

As I have multiple radios in pieces and need a debugger attached to all of them , I just unlock using the JLink Edu then use the STM32 based Jlink which I make from the cheap BluePill boards

redfast00, 2020-05-08:
Okay, thank you for the answer. I just ordered a J-Link edu mini, so we'll see if that works out.

closed 2020-05-08 by rogerclarkmelbourne