rogerclarkmelbourne/OpenGD77 #601: New firmwareloader crash

sm7tix, 2020-05-08:
Hi.
I tried the new firmwareloader and got an error. Sorry for the Swedish but maybe you get a clue whats wrong. Windows 10 pro updated. 

C:\Users\Stefan\Downloads>FirmwareLoader.exe

Ohanterat undantag: System.TypeInitializationException: Typinitieraren för GD77_FirmwareLoader.FirmwareLoader utlöste ett undantag. ---> System.IO.FileNotFoundException: Det går inte att läsa in filen eller sammansättningen LibUsbDotNet, Version=2.2.8.104, Culture=neutral, PublicKeyToken=null eller ett av dess beroenden. Det går inte att hitta filen.
   vid GD77_FirmwareLoader.FirmwareLoader..cctor()
   --- Slut på stackspårning för interna undantag ---
   vid GD77_FirmwareLoader.Program.Main(String[] args)

C:\Users\Stefan\Downloads>

I have also tried the linuxloader with mono and it starts but nothing happens when i hit download and install. No errors in the console at all. I use Linux Mint 19.

f1rmb, 2020-05-08:
Hi,

   Where did you get the Firmware.exe (libusb) version ?
The one using LibUsbDotNet.dll is specifically build for using it with Mono under Linux (you can download the DLL here [https://github.com/rogerclarkmelbourne/OpenGD77/blob/master/tools/firmware_loader/LibUsbDotNet.dll](url)

Cheers.

sm7tix, 2020-05-08:
I downloaded both the windows loader and the linux loader from here
https://github.com/rogerclarkmelbourne/OpenGD77/releases/tag/D20200507
The linux-zip includes the dll. I unpack in one dir and run mono. See screenshot. Starts but nothing happens when i click download and install. I can choose radio. 
[image]

Not use to run mono but i have used it before. Shall the dll be in a specific place?

What are I missing in the windows version?
kind regards
Stefan

f1rmb, 2020-05-08:
Are you sure it's currently started by Mono and not Wine ?

The easiest way is to unzip the archive of the loader somewhere, and run it from a terminal:
`mono FirmwareLoader-Linux.exe`
and since the LibUsbDotNet.dll will be located in the same directory, so you won't have any problem with that.

The difference between Windows and Mono builds is the USBHID communication layer.

Cheers.
...
Daniel

f1rmb, 2020-05-08:
I can (barely) see on the super huge image the download & install button is down.
So yeah, start it from a terminal, I guess there are error messages.

sm7tix, 2020-05-08:
Ok. The last attempt was Wine but no i have used mono again and tried to make better pictures. I  choose gd-77 as tadio and it is connected in download mode, then i click download and install and nothing happens, both in shell and in program. 
It is no big deal but i would like to able to try the dev firmware also. 
Why does thewindows version also crash on my win 10 pro computer? Error msg in my first text. 
[image]
[image]

f1rmb, 2020-05-08:
Okay.
Which mono version is installed on your Mint ?
`mono --version`

About the error message in you first post, it's was related to the missing LibUsbDotNet.dll, but now in your screen capture, I don't see any error message, that's weird.
Under Linux, LibUsbDotNet uses libusb-1.0, is it installed ? 
`apt list --installed | grep 'libusb-1'`

Cheers.
...
Daniel

f1rmb, 2020-05-08:
For the crash under Windows, is it the one related to LibUsbDotNet missing DLL ? In this case, there is something wrong as the Windows version don't use that DLL on this OS. 
I will check the FirmwareLoader.exe from the latest development release.

Cheers.
...
Daniel

f1rmb, 2020-05-08:
Indeed the binary version of the Windows build is wrong, I just raised a PR #603 
Meantime, you can grab the version here : [https://github.com/rogerclarkmelbourne/OpenGD77/blob/c47bb8d7f549d2d2e3bc20fe7ec7d87985fbf966/firmware/tools/FirmwareLoader.exe](url)

sm7tix, 2020-05-08:
> Indeed the binary version of the Windows build is wrong, I just raised a PR #603
> 
> **EDIT**: Okay, Roger updated the binary, so you need to download it again from latest development release page.
Thanks. I will try later today.

sm7tix, 2020-05-08:
Here is a screenshot from the verison question on mono and usblib.
[image]

f1rmb, 2020-05-08:
Okay. Maybe it's because the Mono version is an outdated one. I'm using 6.10.0.104 (but it's nightly builds). I may suggest you to go there and add the [here](
https://www.mono-project.com/download/stable/#download-lin) and follow the instruction to add the latest stable package reposity.

Cheers.
...
Daniel

f1rmb, 2020-05-08:
Another possibility is to use the Python flasher, you just need to install few packages to make it working (yeah, no GUI at all, everything is command line).

Cheers.
...
Daniel

sm7tix, 2020-05-08:
> Another possibility is to use the Python flasher, you just need to install few packages to make it working (yeah, no GUI at all, everything is command line).
> 
> Cheers.
> ...
> Daniel

That i would like :) I use Python every day. How do I download it and what requirements? Can i get the dev branch from there also?

sm7tix, 2020-05-08:
> Okay. Maybe it's because the Mono version is an outdated one. I'm using 6.10.0.104 (but it's nightly builds). I may suggest you to go [here](https://www.mono-project.com/download/stable/#download-lin) and follow the instructions to add the latest stable package reposity.
> 
> Cheers.
> ...
> Daniel

Better now but no choise for the dev firmware that i hoped for. 
[image]

sm7tix, 2020-05-08:
Running the linuxloader and choose gd77 sql fil give this result.
tefan@stefan-MS-7681:~/gdd77$ mono FirmwareLoader-Linux.exe 
 - GD-77 Support
Device Not Found [0x15a2:0x0073].
Error. Can't connect to the GD-77
dmesg sais:
2730.547827] usb 2-1.1: new full-speed USB device number 14 using ehci-pci
[ 2730.657984] usb 2-1.1: New USB device found, idVendor=15a2, idProduct=0073
[ 2730.657987] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2730.657989] usb 2-1.1: Product: USB COMPOSITE DEVICE
[ 2730.657991] usb 2-1.1: Manufacturer: FREESCALE SEMICONDUCTOR INC.
[ 2730.659691] hid-generic 0003:15A2:0073.000A: hiddev2,hidraw5: USB HID v1.00 Device [FREESCALE SEMICONDUCTOR INC. USB COMPOSITE DEVICE] on usb-0000:00:1d.0-1.1/input0
stefan@stefan-MS-7681:~/gdd77$

f1rmb, 2020-05-08:
Okay, so, the Linux version in the release page is not the right version, indeed...
You can grab the latest from the development branch [here](https://github.com/rogerclarkmelbourne/OpenGD77/blob/development/firmware/tools/FirmwareLoader-Linux.exe)

About the USBHID access, have you installed the udev rules  (see one of my post above) ?

About the Python script, yeah, you could get it from the same development branch [here](https://github.com/rogerclarkmelbourne/OpenGD77/tree/development/tools/Python/FirmwareLoader)

Cheers.
...
Daniel

sm7tix, 2020-05-08:
Sory my eye missed the udev rules. In place now and mono version is working for me. Thanks!

f1rmb, 2020-05-08:
That's cool 😉

sm7tix, 2020-05-08:
Python firmware loader works for me also. GOOD. To bad we need to put the radio in download mode otherwise it could be upgraded from for example pi-star. Still good now because i can do it from the raspberry pi  directly now. 
You can close this issue now if you know that the windows version works also. I am a little allergic to Windows... Feel fine in Linux and now with python i feel great :dancing_men:

f1rmb, 2020-05-08:
You can flash from a pi-star.
Do easily do that, you can grab a script I wrote to stop all the pi-star services, then flash, ten restart the services.

Cheers.
...
Daniel

sm7tix, 2020-05-08:
> You can flash from a pi-star.
> To easily do that, you can grab a script I wrote to stop all the pi-star services, then flash, ten restart the services.
> 
> https://github.com/f1rmb/Pi-Star_Binaries_sbin
> 
> Just get the pistar-services script (use 'fullstop' argument, it prevent the pi-star's watchdog to restart the services in the background.
> 
> Cheers.
> ...
> Daniel

Sounds great! Can you explain a little more? I have downloaded the pistar-services. Shall it be enabled with systemctl enable? And then when used full stop i can set the radio in firmware mode and run the pythonflasher?

rogerclarkmelbourne, 2020-05-08:
I've updated the firmware loader linux version in the latest Development release

Please let me know if this is now OK

f1rmb, 2020-05-09:
About the script, just copy it to /home/pi or /root, make it executable
`chmod +x pistar-services`

It just stop cleanly all digital related software, like MMDVMHost, DMRGateway, etc...
`./pistar-services {start|stop|fullstop|restart}`

So, once you put the udev rules on your Pi-Star, installed the Python flasher dependencies, just copy and use the flasher script as you did on your Mint computer, but before flashing it, stop the services, e.g:

```
sudo su
cd /root
./pistar-services fullstop

```

`./gd-77_firmware_loader-hid.py -m GD-77 -d 
`
once your transceiver is flashed,

`./pistar-services start
`

Your done.

Cheers.
---
Daniel

rogerclarkmelbourne, 2020-05-10:
I think this is now fixed

closed 2020-05-10 by rogerclarkmelbourne