rogerclarkmelbourne/OpenGD77 #602: Calibration table address for the DM-5R / RD-5R is incorrect

rogerclarkmelbourne, 2020-05-08:
We currently use the the calibration data from 0x00017c00 , but this seems to have the wrong value for Reference Oscillator frequency.

However, I looked in my Flash backup for memory which looks like calibration data, and I found 6 different copies of the calibration data in the Flash memory !!!

For anyone interested in this, if you open the backup using a hex editor like Hxd ( https://mh-nexus.de/en/hxd/ ) and search for Hex bytes A0 0F in the pattern I find data blocks starting with A0 0F at address

0xF000 (this is the OpenGD77 copy of the calibration)
0x17C00 (this is the address we copy the data from - into 0xF000

But there are also blocks at
0x2F000
0x30000 (same as in 0x2F000)
0x38000 (same as in 0x17C00)
0x4F400 (same as in 0x30000 and 0x2F000)

0x6D000 (same as 0x8F000)
0x8F000 (same as 0x6D000)

So there are 3 completely different sets of calibration data in the Flash memory

One version is in

Version 1 0x17C00 and 0x38000
Version 2 0x2F000 and 0x30000 and 0x4F400
Version 3 0x6D000 and 0x8F000

We already use the version at 0x17C000 which is definitely wrong.

So the correct version must be version 2 or version 3. I will need to try both of these versions, to see if either of them gives the correct frequency.

Looking at the data in my backup, the data in 0x6D000 and 0x8F000 seem to have values for the Reference oscillator which look like they are correct, because the value needed by my radio is around 975 and all the other data have values more than 1000 and in some cases more than the max value of 1023

I will try using 0x6D000 in my radio.

PanKaczka, 2020-05-08:
Some time ago I've messed my RD-5R calibration.
As far as I remember I've used calibration option from the readout application to modify the RSSI reading, it turned out that in EEPROM it can only toggle bits one way as it probably was not doing the erase procedure.
Nevertheless I was trying to restore the calibration data, in the end I've hooked a programmer directly to eeprom chip and found working copy under 0x8F000. In my case the calibration in use was also taken 0x6D000.

I'm attaching the EEPROM with my calibration data (also with updated font for original FW), maybe it can be useful.
[rd5r_eeprom.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/4599402/rd5r_eeprom.zip)

rogerclarkmelbourne, 2020-05-08:
This is already fixed in the development version, including the power.

See my post on the OpenGD77 forum about 9 hours ago

closed 2020-05-08 by rogerclarkmelbourne