rogerclarkmelbourne/OpenGD77 #604: License conflict

d235j, 2020-05-08:
The GNU GPL v2, section 6 does not permit adding additional terms that further restrict the code, such as

> Use of this software for commercial purposes is strictly forbidden.

Why is this term present? Is there code in this repository that is from other sources, not under the GPLv2? If so, other approaches must be taken for compatibility — adding terms that do not cause restriction on the GPLv2 code is one such approach.

For instance, if someone wants to take snippets of the code and import them into another project that is under the GPLv2, the GPLv2 permits this with attribution. Restricting "commercial use", which can have a wide range of definitions (e.g. accepting contributions could be considered "commercial use"), is prohibited under the GPLv2, **and makes the software non free and open source**.

I have noticed this term on other amateur radio projects, with the maintainers refusing to discuss the situation. It should be understood that such code is likely not permitted to be distributed, or perhaps even used.

d235j, 2020-05-08:
Please see here for a detailed discussion — that term is likely meaningless.

> The second sentence of GPLv2 §6 does the opposite; it bounds from the top. It prohibits any licensor along the distribution chain from placing additional restrictions on the user. In other words, no additional requirements may trump the rights and freedoms given by GPLv2.

https://copyleft.org/guide/comprehensive-gpl-guidech8.html#x11-530007.3

If there is code here that is in conflict with the GPL, that needs to be handled in other ways than adding useless terms.

rogerclarkmelbourne, 2020-05-08:
I don’t think I can change this.

The code is a fork of the code from the HamDV repo, owned by Kai DG4KLU.

rogerclarkmelbourne, 2020-05-08:
BTW.

I think the reason that this license and additional terms were used, because its the same as used by the MMDVMHost project

See

https://github.com/g4klx/MMDVMHost

Its also GPL 2.0 and includes the text

"Use of this software for commercial purposes is strictly forbidden."

And from what we understand this has not been a problem for that project.

d235j, 2020-05-08:
@rogerclarkmelbourne I can open an issue on the upstream repo.

Regarding MMDVM/MMDVMHost: people have brought it up there too, and the maintainer adamantly refuses to address it: https://github.com/g4klx/MMDVM/issues/22 (and others), and https://github.com/g4klx/MMDVMHost/issues/373

That project has a separate license issue: it includes source that is under non GPL compatible licenses without an exception.

It is "not a problem" in the sense that people using it probably don't care what the license is, but it definitely cannot be considered "free and open source" if that term is to be respected.

rogerclarkmelbourne, 2020-05-08:
Well, this is outside of my control, as I can't change this in my fork.

So I'll have to close this issue.

Please feel free to open an issue upstream and perhaps the owner of the repo will respond

closed 2020-05-08 by rogerclarkmelbourne

redfast00, 2020-05-14:
@rogerclarkmelbourne I think these extra restrictions are meaningless anyway, so they don't really apply. The reason I'd like them to be gone is to be able to say "my radio is running Free and Open Source firmware", without having to worry about the licence. @talentraspel has removed the issue-section on their repo (and that's okay, sometimes opensource maintainers get burned out from dealing with issues), so opening an issue there and asking them to remove these restrictions is not really possible. I think an appropriate compromise would be to remove the (useless) restriction in the README and other locations, but keep it in the folder that has the MMDVMHost source code. If people then absolutely want to have clean-licenced code, they can rewrite that part from scatch. I'm willing to create a PR that does this.

rogerclarkmelbourne, 2020-05-14:
I can't change this, because its the license from the original author of the code.

The original author imposed limitations he wanted on the use of the firmware, and I'm abiding by the wishes as stated in his readme.

talentraspel, 2020-05-15:
I'm sorry but the licensing will stay the way it is and I won't hire a lawyer on a free hobby project to make it more clear so the current wording has to be sufficient.

It is my will to prevent any commercial exploitation of the project or the works that use it. Please respect that, thanks a lot.

1) This software is licenced under the GPL v2 and is intended for amateur and educational use only.
2) Use of this software for commercial purposes is strictly forbidden.

It is the same like MMDVM and keeps up the hamspirit by preventing commercial exploitation of the work.

balr0g, 2020-05-15:
The second statement, which attempts to strictly forbid commercial use of any sort, is superseded by the terms of the GPL that prohibit such additional restrictions. I believe I provided enough information to demonstrate as much. Some people even contacted the Free Software Foundation for clarification.

GPL in itself should be sufficient to “keep up the hamspirit”, by requiring people distributing the software to keep it open source.

A statement that says that commercial use is strongly discouraged would be reasonable, but strictly forbidding it flies in the face of the GPL: one cannot follow the terms of the GPL and this additional restriction at the same time.

d235j, 2020-05-15:
The second statement, which attempts to strictly forbid commercial use of any sort, is superseded by the terms of the GPL that prohibit such additional restrictions. I believe I provided enough information to demonstrate as much. Some people even contacted the Free Software Foundation (which has lawyers and which has devised the GPL in the first place) for clarification, and those people were pointed to the interpretation I quoted above.

GPL in itself should be sufficient to “keep up the hamspirit”, by requiring people distributing the software to keep it free and open source.

A statement that says that commercial use is strongly discouraged would be reasonable, but strictly forbidding it flies in the face of the GPL: one cannot follow the terms of the GPL and this additional restriction at the same time.

d235j, 2020-05-15:
To add, software that is under terms that prohibit commercial use cannot be called free and open source:

https://opensource.org/osd-annotated (Part 6)

https://www.gnu.org/philosophy/free-sw.html.en

Therefore this is “open source” unless this limitation on fields of use is dropped.