rogerclarkmelbourne/OpenGD77 #606: "DMR Beep: Both" not working

rjstone, 2020-05-09:
In build May 4 2020 8:54:11[ ada9153 ] for GD77, the "DMR Beep: Both" setting doesn't sound the start beep, only the ending beep.

rogerclarkmelbourne, 2020-05-10:
Thats an old commit, which has nothing to do with the DMR beeps.

And I can't reproduce this bug on my radios.

rogerclarkmelbourne, 2020-05-11:
Closed because of no response from the OP.

rjstone, 2020-05-11:
I was copying the commit off the firmware info screen. I’ll just try the next release and see if it still happens. Not sure what’s going on.

rogerclarkmelbourne, 2020-05-12:
The latest versions are now handled by the Github release system.

I'll remove the old SGL files from the latest commit, 
(Its not practical to remove them from old commits)

rjstone, 2020-05-12:
I got it from here https://github.com/rogerclarkmelbourne/OpenGD77/releases/tag/R20200503 Maybe the firmware info screen is displaying the wrong commit hash? I should have installed the latest pre-release not the latest release but I got that mixed up it seems.

rogerclarkmelbourne, 2020-05-12:
OK. F1RMB just reported to me that the beep does not happen 100% of the time.

I'll investigate

rjstone, 2020-05-12:
(This is KV0A BTW, sorry) Some more useful information: I read my codeplug from the radio using the CPS, then wrote the code plug back to the radio and now the start beep seems to be happening all the time, or at least every time I've tried it. (I can't key up too fast or BM will ban me. Let me shut off my hotspot and try it.....)

Ok, I hit PTT 30 times in quick succession and 7 of those times I didn't get the start beep. So rewriting the CP helped some but there still seems to be a little nondeterministic issue of some sort.

rogerclarkmelbourne, 2020-05-12:
Yep.

Its probably a timing problem and may only effect simplex DMR (including simplex hotspots)

I'm investigating it now

rogerclarkmelbourne, 2020-05-12:
Should be fixed by 
https://github.com/rogerclarkmelbourne/OpenGD77/commit/13d42bb5e5a9c699cf4ed3a773be2cd6436bd1d2

closed 2020-05-12 by rogerclarkmelbourne