rogerclarkmelbourne/OpenGD77 #613: Channel bandwidth doesn't work correctly

AlekseiChilikin, 2020-05-12:
Changing the channel bandwidth doesn't work correctly. 

[image]
Changing the bandwidth parameter to 12.5kHz, GD-77 still continues to work with 25kHz

f1rmb, 2020-05-12:
Hi,

  How do you notice that ?

Cheers.
...
Daniel

AlekseiChilikin, 2020-05-12:
Hi,

In the channel details I set the channel bandwidth to 12.5 kHz
When I start broadcasting. I can hear my radio on two adjacent channels. For example, I start working on the frequency 446.01875. On other radios, I can receive my broadcast on the frequency 446.00625 446.01875 and 446.03125. GD-77 runs on 50mW. Other radios are quite far away.

Aleksei

rogerclarkmelbourne, 2020-05-12:
Did you compare the the same radio running the official firmware with the same settings?

Also, did you do a spectrum plot of some kind e.g. using a spectrum analyser or a SDR dongle or another radio with some sort of analyser display.

AlekseiChilikin, 2020-05-13:
Hi,
I forgot to specify. Channel operation in FM mode. 
Working in DMR mode, the problem is not observed.

The problem also persists on the factory firmware (V4.02.08). In FM mode, GD-77 ignores the channel bandwidth settings. I'll check the spectrum next week.

Have a nice day!

rogerclarkmelbourne, 2020-05-13:
If the radio does the same with the official firmware, then its likely your testing or expectations of bandwidth of the transmission are not correct.

closed 2020-05-14 by rogerclarkmelbourne