rogerclarkmelbourne/OpenGD77 #614: RD-5R talk group memory location possibly wrong.

Jahnkeanater, 2020-05-13:
After flashing the newest development build from stock firmware the talkgroups became corrupt. I did a factory reset and flashed an empty code plug and there is still 1 old talkgroup and 2 blank talk groups with no tittles.

Seems like the radio reads more memory at the beginning of the list than CPS. You also cannot delete the talkgroup on the radio.

Factory reset does not seem to erase the codeplugs properly.

rogerclarkmelbourne, 2020-05-13:
The stock codeplug  format is not 100% compatible with this firmware.

Use the tools here to save the RD-5R codeplug to CSV and then import them into a OpenGD77 compatible codeplug.

http://www.gb3gf.co.uk/downloads.html

I'll update the User Guide to highlight this.

closed 2020-05-13 by rogerclarkmelbourne

rogerclarkmelbourne, 2020-05-13:
User guide has been updated

Jahnkeanater, 2020-05-15:
How do I delete the corrupt code plug that is not shown in CPS?

rogerclarkmelbourne, 2020-05-16:
Github issues are not a support forum

Use https://www.opengd77.com

Jahnkeanater, 2020-05-16:
After a factory reset and after flashing a blank codeplug using CPS should that not erase all talkgroups?

Also does the delete function in the menu just not work yet?