rogerclarkmelbourne/OpenGD77 #622: Switch to newlib nano instead of redlib as C library

redfast00, 2020-05-16:
Currently, redlib is used as the libc in this project. The disadvantage of this is that redlib is proprietary, and no source code is available. This makes the project hard to build without the mcuxpresso IDE (my end goal is to be able to build the entire project without having to install proprietary tools and to document how to do this). Mcuxpresso also offers two other libc's: newlib and newlib nano. I tried to switch the project to use newlib nano, but found out there's just barely not enough space in SRAM_LOWER_BOTTOM (we need another 72 bytes at least).

Redlib:
```
    PROGRAM_FLASH:        476 KB       494 KB     96.36%
       SRAM_UPPER:       27600 B        64 KB     42.11%
SRAM_LOWER_BOTTOM:       27444 B      27488 B     99.84%
  SRAM_LOWER_AMBE:          0 GB       8744 B      0.00%
   SRAM_LOWER_TOP:          8 KB      29304 B     27.96%
```

Newlib nano:

```
Memory region         Used Size  Region Size  %age Used
/usr/local/mcuxpresso-ide/ide/plugins/com.nxp.mcuxpresso.tools.linux_11.1.0.202001081728/tools/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /usr/local/mcuxpresso-ide/ide/plugins/com.nxp.mcuxpresso.tools.linux_11.1.0.202001081728/tools/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libcr_newlib_semihost.a(syscalls.o): in function `_write':
   PROGRAM_FLASH:        476 KB       494 KB     96.36%
      SRAM_UPPER:       27476 B        64 KB     41.93%
SRAM_LOWER_BOTTOM:       27560 B      27488 B    100.26%
 SRAM_LOWER_AMBE:          0 GB       8744 B      0.00%
syscalls.c:(.text._write+0x0): multiple definition of `_write'; ./source/SeggerRTT/Syscalls/SEGGER_RTT_Syscalls_GCC.o:/home/jdv/Projects/OpenGD77-codec2/firmware/Debug_GD-77/../source/SeggerRTT/Syscalls/SEGGER_RTT_Syscalls_GCC.c:110: first defined here
  SRAM_LOWER_TOP:          8 KB      29304 B     27.96%
/usr/local/mcuxpresso-ide/ide/plugins/com.nxp.mcuxpresso.tools.linux_11.1.0.202001081728/tools/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: firmware.axf section `.data' will not fit in region `SRAM_LOWER_BOTTOM'
/usr/local/mcuxpresso-ide/ide/plugins/com.nxp.mcuxpresso.tools.linux_11.1.0.202001081728/tools/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: region `SRAM_LOWER_BOTTOM' overflowed by 72 bytes
collect2: error: ld returned 1 exit status
make: *** [makefile:55: firmware.axf] Error 1
```

I pushed my changes on https://github.com/redfast00/OpenGD77/tree/newlib-instead-of-redlib, so you can check it out if you want to.

redfast00, 2020-05-16:
It seems that some other people have tried this as well: http://www.opengd77.com/viewtopic.php?f=16&t=835&sid=d05cfa1a1539908d2e3db6ee19cfba1f

rogerclarkmelbourne, 2020-05-16:
Using this lib does not seems to save space in the ROM and actually appears to overflow one of the RAM sections

`SRAM_LOWER_BOTTOM:       27560 B      27488 B    100.26%`

Unless you intend to **not** support DMR in your version, this seems to be pointless, because AMBE used by DMR is proprietary, with no source code for the codec, hence we have to use the binary sections from the official firmware for the AMBE codec.

See  
https://github.com/rogerclarkmelbourne/OpenGD77/tree/master/firmware/linkerdata

redfast00, 2020-05-16:
That's true, the current configuration seems to overflow one of the sections. However, in the other config, the `SRAM_LOWER_BOTTOM` seems to be measured out almost exactly for the space that is needed, having exactly 4 bytes left. The newlib only uses 76 bytes more than redlib. On the thread I linked, they expanded this section, but then had problems with the calibration data, which may or may not be related. Is there some data that is needed directly after the SRAM_LOWER_BOTTOM?

redfast00, 2020-05-16:
For your convenience, I made a git commit from the changes in the forum post: https://github.com/rogerclarkmelbourne/OpenGD77/compare/master...redfast00:make-on-linux?expand=1

Nimayer, 2020-05-24:
Hi @redfast00, thank you for your effort, being able to compile OpenGD77 under Linux using an open source libc would be great.

I've dug deeper in the problem IU2KIN and others faced in [this attempt](http://www.opengd77.com/viewtopic.php?f=16&t=835&sid=d05cfa1a1539908d2e3db6ee19cfba1f).
The source archive downloadable [here](https://iu2nro.it/files/OpenGD77.zip) includes a Makefile for building on Linux with `arm-none-eabi-gcc` and `gnu-newlib-nano`.
It should also include a modified linker script to solve the SRAM size issue.

The problem is that OpenGD77 compiled using this source doesn't fully boot up, it gets stuck in reading calibration data
[image]

I've dumped the flash (using CPS community edition) of the radio running the 20200503 release and the version compiled with Makefile, and it appears the flash content is not read properly in the makefile version, this causes the error above.

Here is an example of a portion of the flash dump
[image]
I am suspecting there is some issue in the `SPI_Flash_read()` function due to the build system change

rogerclarkmelbourne, 2020-07-05:
I'm closing this issue, as anyone who wants to build this using newlib nano can do so using the Makefile version

PR #635 (2020-06-28): Add Newlib support
Added the support to build with newlib instead of redlib.
I also added a makefile to build the project from Linux without the NXP IDE which supports both newlib and redlib
This should not break the redlib build from MCUXpresso however it is still not possible to compile with the newlib from the IDE
Closes #622
closed 2020-07-05 by rogerclarkmelbourne