rogerclarkmelbourne/OpenGD77 #638: Choppy sound on when using local repeater

sm7tix, 2020-05-26:
I am setting up a new repeater with a Kenwood TKR-751 using MMDVM and Pi-star. When I trim it to work with DMR and D-star my GD-77 give me choppy sound but my Retevis RT3S works great. If i use my GD-77 with my hotspots there is no problem. The "choppyness" dont sound like bit error or so but more like the bandwith is to tight or other hardware issue. The repeater uses wide4k for the bandwith and if i use narrow the DMR do not decode at all. What do you think this is? Can the radio be calibrated to work? I am good in Linux but not radiohardware :D 
I put in one question also, is it possible to get voice speach on the GD-77? I follow you here and i have seen descussions about it, but maybe just on gd-77s?
Otherwise i love my radio GD-77.

na7q, 2020-05-28:
This sounds exactly like the issue I have. Are you using a cheap POG style MMDVM board? Or some other board? Is it on a regular Pi or a Pi Zero? 

Here's what I have noticed, there are 2 types of cheap MMDVM boards available. Both can cause the chop to OpenGD77 radios. I'd say the issue is caused by multiple things. The board, the Pi, and or the radio, or a combo of it all. With one of my boards, you can use it on any Pi without choppy audio on OpenGD77. The other board, you can use it on a Pi Zero with no issues at all. If you try it on a Pi 3 B+, the audio is choppy. Even though things appear properly tuned, it works on all radios, except OpenGD77.

My suggestion is to avoid the POG boards and to use a Pi Zero if you have one of the newer models of the board. That should clear things up.

sm7tix, 2020-05-28:
> This sounds exactly like the issue I have. Are you using a cheap POG style MMDVM board? Or some other board? Is it on a regular Pi or a Pi Zero?
>
> Here's what I have noticed, there are 2 types of [...]

HI
I am using a RepeaterBuilder board v2 and have adjusted the deviation so it works fine on a Retevis RT3S and a Anytone d578uv. The D-star is also working fine on that card. The Raspbian is a Pi 3 Model B (1GB). So you think a Pi Zero should work better?

na7q, 2020-05-28:
I have not yet used a RepeaterBuild board. But yes, in my experience with my 4 MMDVM repeaters, switching from the Pi 3's to a Pi Zero completely resolved the issue. Given the board was a newer generation. I think the RepeaterBuilder boards would work better on the Pi Zero W. You won't know until you try it however.

sm7tix, 2020-05-28:
Tried a Pi Zero W and it sounds the same choppy sound on GD77 with latest dev fw. Also the latest Pi-star but i had same on 4.1.1.

na7q, 2020-06-01:
I forgot to add some small details. What server are you using? I noticed that if I connect to Brandmeister, I have no issues at all. If I connect to my HBlink server, it's 100% choppy at all times with the one board.

sm7tix, 2020-06-01:
I am uding Brand Meister with pi-star snd Repeaterbuilder card V2. Dmr and 
D-star active. The repeater is a Kenwood tkr-751 eith mode wide4k. The 
chopping sounds lika Agc or something cut the signal. Not so much digital 
choppyness.
Med vänlig hälsning

Stefan Rydberg
0767-882514

rogerclarkmelbourne, 2020-06-02:
Did you try reloading the official firmware onto the radio and confirming that it does not have the problem.

na7q, 2020-06-02:
Sounds identical to what I see. Can you record a video of it?

Roger, I have the issue only with OpenGD77. Reloading the official firmware resolves it for me.

sm7tix, 2020-06-02:
Video. Hmm 10MB only. The video become 129MB. How to compress?

sm7tix, 2020-06-02:
I manage to make a smaller video of the chopping sound.
[sm7tix.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/4718503/sm7tix.zip)

sm7tix, 2020-06-02:
Clicked wrong...

sm7tix, 2020-06-02:
I shouyld buy me some new eye...

sm7tix, 2020-06-02:
I have uploaded a video. 

---
Med vänlig hälsning

Stefan Rydberg
0767-882514 

2020-06-02 07:20 skrev Mike:

> Sounds identical to what I see. Can you record a video of it? 
> 
> Roger, I have the issue only with OpenGD77. Reloading the official firmware resolves it for me. 
 

Links:
------
[1]
https://github.com/rogerclarkmelbourne/OpenGD77/issues/638#issuecomment-637282588
[2]
https://github.com/

na7q, 2020-06-12:
As to your video. My experience is identical, only that mine is much faster at the chop. I'm noticing extra chop while mobile again on my local Motorola repeater. This radio is a tough job!

sm7tix, 2020-06-12:
Ok. I hope they find a fix for it.
Med vänlig hälsning

Stefan Rydberg
0767-882514

rogerclarkmelbourne, 2020-06-14:
Looks like you have CC filtering disabled.

Did you try putting the filtering back to the normal settings.

Disabling the CC filtering used to determine which CC a station is transmitting on. Its not intended to be used all the time, because its basically a hack as the C6000 DMR DSP chip is only designed to operate on a defined CC

sm7tix, 2020-06-14:
I have tried both.
For the moment the repeater is unmounted and waiting to be transported to 
the Qth. I will do more testing then. I have no timeframe though. Maybe a 
few weeks.

Med vänlig hälsning

Stefan Rydberg
0767-882514

rogerclarkmelbourne, 2020-08-20:
Can't replicate, so I will need to close this until more information is available

closed 2020-08-20 by rogerclarkmelbourne