rogerclarkmelbourne/OpenGD77 #640: Selcall-tone (2 tone)

djshadowxm82, 2020-06-03:
So did some looking in the datasheet for the HR_C6000 and found a section For 2-tone decode and transmit in the 6000. I realize the AT-1846 is setup to do FM analog. but I think its also capable of handling some of the FM tasks such as 2/5 tone. Was wondering if we could possibly investigate the plausibility of at least supporting 2-tone decode (Use this as a fire radio as well. Would be handy to have decode of firecalls)

A2.2.4 Selcall-tone (2-tone)
Configure 0xa1[7:0] = 8'h10 to enter 2-tone mode; 

A2.2.4.1 Parameter configuration and initialization A2.2.4.1 Parameter configuration and initialization

 • Configure 0xa0[7:0] = 8'h39, set 2-tone transmit modulation frequency deviation to 1.8KHz; 
 • Configure 0xa4[7:0] = 8'h05, set the 2-tone first tone transmission time, default 500ms, adjustable step length is 100ms;
 • Configure 0xa3[7:0] = 8'h05, set 2-tone adjacent frame gap duration, default 500ms, adjustable step length is 100ms
 • Configure 0xa5[7:0] = 8'h05, set the 2-tone second tone transmission time, default 500ms, adjustable step length is 100ms;
 • Configure 0xa6[7:0] = 8'h0a, set 2-tone long tone transmission time, default 1000ms, adjustable step length is 100ms;
 • Configure 0xa7[7:0] = 8'h0a to set the 2-tone demodulation threshold; configure 0xa7[7:0] = 8'h0a to set the 2-tone demodulation threshold; 
 • Configure {0xaf[3:0] , 0xae[7:4]}, set the 2-tone sending code, which is composed of “A ”, “B ”,“ C ”, “D” paired or output separately as a long tone;

Receive demodulation coefficient configuration: Configure {0x104, 0x103, 0x102}=24'h767041 (A tone 1981Hz corresponding demodulation value); 
Receive demodulation coefficient configuration: Configure {0x107, 0x106, 0x105}=24'h7ce537 (B tone 1124Hz corresponds to the demodulation value); 
Receive demodulation coefficient configuration: Configure {0x10a, 0x109, 0x108}=24'h7c7b1e (C tone 1197Hz corresponding to demodulation value);
Receive demodulation coefficient configuration: Configure {0x10d, 0x10c, 0x10b}=24'h7c0285 (D tone 1275Hz corresponds to demodulation value); 

Refers to sending: When PTT is valid, the system will save the 2-tone code stored in registers 0xaf[3:0], 0xae[7:4] to send. The transmission format is specified by 0xa3 to 0xa6. 2-tone is single mode, that is, the sequence code is sent only once at a time, and the next time. Need to wait until the new PTT is valid;

Receive: The MCU temporarily compares the data in register 0x93 with the two-tone receive address preset by the software at each FM interrupt, and receives the register 0x94[3:0]. The data is compared with the two-tone reception address preset by the software, and register 0x94[3:0] is received. The second FM interrupt comes to temporarily compare the data in register 0x93 with the two-tone receive address preset by the software, and compare the data in register 0x94[3:0] with the long-tone receive address preset by the software. If there is a match, Then open the speaker channel (0x36[7:0] = 8'h72 ), otherwise turn off the speaker (0x36[7:0] otherwise turn off the speaker (0x36[7:0] = 8'h80 ).

rogerclarkmelbourne, 2020-06-04:
My time is already 100% committed on other functionality for the OpenGD77, specifically the voice prompts for visually impaired operators.

Please submit a PR for this when you get it working, and it will be considered for integration, as long as it does not impact on any existing functionality.

rogerclarkmelbourne, 2020-06-07:
Closing because of no response from OP within 48 hour

closed 2020-06-07 by rogerclarkmelbourne