rogerclarkmelbourne/OpenGD77 #641: Baofeng DM-1701

v0l, 2020-06-05:
Hi,

I recently purchased this radio and was wondering what the process might be to port this firmware to this device?

When connected i get usb interfaces listed as below:

```
Bus 001 Device 012: ID 0483:df11 STMicroelectronics STM Device in DFU Mode
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0483 STMicroelectronics
  idProduct          0xdf11 STM Device in DFU Mode
  bcdDevice            2.00
  iManufacturer           1 AnyRoad Technology
  iProduct                2 Digital Radio in USB mode
  iSerial                 3 00000000010C
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x002d
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          2 Digital Radio in USB mode
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       254 Application Specific Interface
      bInterfaceSubClass      1 Device Firmware Update
      bInterfaceProtocol      2 
      iInterface              6 @Internal Flash   /0x08000000/03*016Kg
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           0
      bInterfaceClass       254 Application Specific Interface
      bInterfaceSubClass      1 Device Firmware Update
      bInterfaceProtocol      2 
      iInterface              7 @1G Flash Memory /0x10000000/512*256Kg
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       2
      bNumEndpoints           0
      bInterfaceClass       254 Application Specific Interface
      bInterfaceSubClass      1 Device Firmware Update
      bInterfaceProtocol      2 
      iInterface              8 @SPI Flash Memory /0x00000000/16*064Kg
      Device Firmware Upgrade Interface Descriptor:
        bLength                             9
        bDescriptorType                    33
        bmAttributes                       11
          Will Detach
          Manifestation Intolerant
          Upload Supported
          Download Supported
        wDetachTimeout                    255 milliseconds
        wTransferSize                    1024 bytes
        bcdDFUVersion                   1.1a
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

rogerclarkmelbourne, 2020-06-05:
That radio appears to use a STM32 processor.

This firmware is only for the NXP MK22 processor.

Try raising an issue on this repo, which is attempting to port this firmware to STM32 based radios

https://github.com/n1zzo/OpenDMR

v0l, 2020-06-05:
Ok thanks, will check it out!

closed 2020-06-05 by rogerclarkmelbourne

Marczydlo, 2020-12-09:
Newest BAofeng DM-760  IS POSSIBLE TO USE OPen GD-77?