rogerclarkmelbourne/OpenGD77 #645: Mandatory TS setting not applied when selecting TG from list

NeuerUser, 2020-06-13:
Reproduce:
- set one contact TG to mandatory TS2 (example)
- when selecting the TG with the left and right key, the TS is changed accordingly
- now press three times "#", then select the TG
- now the mandatory TS setting is not applied

rogerclarkmelbourne, 2020-06-14:
OK.

I can replicate this bug.

rogerclarkmelbourne, 2020-06-14:
Interesting thing, is that I think there is another bug with the TS override, because the Channel TS override is supposed to be separate from the VFO TS override and the code is no handling this correctly either.

Umm. I'll need to fix one bug at a time ;-)

NeuerUser, 2020-07-21:
The bug seems to be fixed in the latest dev release. Thanks!

There is still some small TS bug left:

- turn on the phone on a DMR channel. Some TG is selected and the correct TS is shown (e.g. TG1 and TS1)
- now temporarily choose another TG with a different TS (e.g. TS228 and TS2), either from contact list or via ###
- the new TG is chosen with the correct TS
- now turn off the TX and on again: the TG reverts to the stored one, but not the TS (in this example TG1, but TS2 stays)

Do you prefer to have a new issue opened for this?

rogerclarkmelbourne, 2020-07-21:
Cant replicate

What TS do you have assigned to the Channel, and does the Contact have a TS override.

NeuerUser, 2020-07-21:
Contact has TS override active (that's what I meant with "Mandatory TS setting"). In my specific example that I use:

- Channel has TG22801 with TS override set to TS1
- Contact I choose is TG9990 with TS override set to TS2
- When I turn on the TX, channel selected with TG22801 and TS1 override (correct)
- turn off, turn on, still the same
- now choose contact TG9990 with TS2 override => TG9990 and TS2 is set (correct)
- turn off TX, turn on again, now contact reverts to TG22801, but TS stays on TS2 (incorrect)

Hope this helps, otherwise I can post a short video.

NeuerUser, 2020-07-21:
OK, I maybe found an important detail:

The contact only reverts, if the new contact chosen via ### is a private contact, not a talk group.

rogerclarkmelbourne, 2020-07-21:
The whole contact selection function needs to be overhauled.

There is a plan to do this, but no one is free to do it at the moment

rogerclarkmelbourne, 2020-07-21:
See https://github.com/rogerclarkmelbourne/OpenGD77/issues/647

NeuerUser, 2020-07-22:
> The whole contact selection function needs to be overhauled.
> 
> There is a plan to do this, but no one is free to do it at the moment

Thanks for the info. I think this is a minor bug, so there is no urgency at all.

f1rmb, 2021-03-06:
I think this one can be closed, TS management was fixed some times ago (and I just retested it).

closed 2021-04-20 by rogerclarkmelbourne