rogerclarkmelbourne/OpenGD77 #646: Many settings in options are reset when uploading DMR database

NeuerUser, 2020-06-13:
When using the DMR ID upload function, the handset must be turned on together with the blue key, the green key and the star key in order to enter "Memory access mode".
This, however, seems to reset many setting, e.g. transmit power, all display and sound options.

Is there any way to achieve a DMR ID upload without resetting these settings?

rogerclarkmelbourne, 2020-06-13:
Memory Access Mode is a feature in the official Radioddity firmware, the OpenGD77 firmware uses a completely different data protocol supported by the OpenGD77CPS

closed 2020-06-13 by rogerclarkmelbourne

NeuerUser, 2020-06-13:
Ok, thanks for the info. 

The OpenGD77CPS , however, told me to enter this memory access mode and also specified three keys to press.

Reproduce: 
remove handset, open CPS, go to DMR ID, select write (without handset), then the message appears to switch the handset to memory access mode. (That's why I did this)

rogerclarkmelbourne, 2020-06-13:
Thats because its switching to legacy support. 

The bug is that the OpenGD77CPS still has some legacy code for the Official firmware, which need to be removed. 

The message is erroneous

This is not a bug in the firmware.

rogerclarkmelbourne, 2020-06-13:
I have pushed a fix for this to the CPS, but I don't have time to rebuild and create an installer.

Please rebuild the OpenGD77CPS yourself if you need the latest version

NeuerUser, 2020-06-13:
Thanks a lot for your great work. This is really a masterpiece of software and I am so glad to follow the development of this.