rogerclarkmelbourne/OpenGD77 #648: RD-5R - Sound configuration is reset every power cycle

EkosFox, 2020-06-14:
Sorry if is replicate.
If I disable Prompt beep, after turning the radio off and on again it doesn't remember the configuration.

Replicate:
* Go to Menu > Sound options. 
* Turn Prompt to "Silent" and click Menu button to confirm
* Turn off HT
* Turn HT back on
* The option is back at stock

rogerclarkmelbourne, 2020-06-24:
Can't replicate

rogerclarkmelbourne, 2020-06-25:
Closed because bug can't be replicated and OP did not respond

closed 2020-06-25 by rogerclarkmelbourne

gigawatts, 2020-07-20:
@rogerclarkmelbourne I can confirm that following @EkosFox steps does indeed cause the prompt option to restore to "Beep".

rogerclarkmelbourne, 2020-07-20:
Try the development version. A similar bug was fixed last week

gigawatts, 2020-07-20:
Sure, I can give it a shot, you are referring to the June 28th "pre-release" ?
Tag: D2020.06.28  Hash: fed4a21

rogerclarkmelbourne, 2020-07-20:
Yes. It fixed on sound related problem