rogerclarkmelbourne/OpenGD77 #650: TGID ONLY Filtering on DM1801 (enhancement)

ssoespre, 2020-06-16:
It will be nice, if possible, to turn on TG Only filtering on DM1801, 
Maybe it's not for everybody, but it is nice to have the option to unmute audio only for th TGID we filter.

Many thanks :)

rogerclarkmelbourne, 2020-06-16:
All versions of the firmware have the same features

ssoespre, 2020-06-16:
Yes. As for now, i can turn on filter TGID only if CC and TS also being turned on. Is it possible in the future update, we can select the filters individually?

These features are so great.

rogerclarkmelbourne, 2020-06-16:
OK. So this feature request has nothing to do with the DM-1801

The filters are already too complicated for most people to use, so adding a way to turning them on and off individually won't be done by me.

You'll need to code your own custom version to do this

closed 2020-06-16 by rogerclarkmelbourne

ssoespre, 2020-06-16:
Noted. Thankyou.

rogerclarkmelbourne, 2020-10-04:
I think this is now supported

ssoespre, 2020-10-04:
yes mate, it is now supported.
many thanks for all the efforts to make these firmwares better and better everytime, cheers.

rogerclarkmelbourne, 2020-10-04:
OK. Cool.

I just thought I better review anything still tagged as an enhancement