rogerclarkmelbourne/OpenGD77 #651: Zone Skip & All Skip in Channel Details

ssoespre, 2020-06-18:
I just found out that changes upon Zone Skip & All Skip in Channel Details were saved a while, but then reset after a few minutes.

f1rmb, 2020-06-18:
Did you saved with <BLUE> + <GREEN> ?
<GREEN> only is temporary until you change to another channel.

ssoespre, 2020-06-22:
i have tried Blue + Green to save the Zone/All Skip setting, it was saved but only until you cycle the power and it was reset.

f1rmb, 2020-06-22:
What kind of transceiver?

ssoespre, 2020-06-22:
DM1801...sorry not mentioning previously

f1rmb, 2020-06-22:
Okay, I'll check that. Thanks.

rogerclarkmelbourne, 2020-06-22:
Worked fine for me on my DM1801

Set both to both Skip to Yes.
Press SK2 + Menu to save
changed channel
changed to VFO and back
power cycled etc

The settings did as I expected

Only retained if I change channel if I press SK2 + Menu to save

Otherwise they only remain if I stay on the same channel

Which is what the firmware is supposed to do.

It does not keep a cache of all the channel data for every channel in the zone
Only for the current channel.

It does retain it in memory if you goto the VFO and back to the same channel, because it caches the Current Channel, and also the current data for VFO A and VFO B.

If you still don't think its working like this. You'll need to post a list of steps to reproduce the problem.

ssoespre, 2020-06-22:
My BAD. i forgot to press SK2 previously.
It is working all good now. 
Many thanks for the enlightments. Cheers.

closed 2020-06-22 by ssoespre