rogerclarkmelbourne/OpenGD77 #672: Radio Exiting Hotspot Mode When Transmitting

aminhusni, 2020-07-09:
Radio: Baofeng DM-1801
Firmware: 30f033f
Pi-Star: 4.1.2
Modem Mode: OpenGD77:v0.1.4

I am encountering this problem where when at random, there is traffic coming from the internet, the radio attempts to transmit and it would just exit out of hotspot mode. 
Pi-Star would then be stuck at transmitting mode (the TX Second indicator is not frozen)

[image]

f1rmb, 2020-07-09:
It's a well known problem when too much RF goes back to the PI. In that case, the kernel USB driver reset the bus.
[https://www.opengd77.com/viewtopic.php?f=11&t=82](url)

aminhusni, 2020-07-10:
Thank you. 
I had attached two ferrite beads to my USB cable and it seemed to has solved the problem so far. 

[image]

closed 2020-07-10 by aminhusni

rogerclarkmelbourne, 2020-07-10:
It will eventually crash the USB unless you use an external antenna. 

You can't transmit RF that close to unshielded devices without inducing glitches in the USB

mti-sk, 2020-07-22:
A new issue has been added, with a similar problem :
https://github.com/rogerclarkmelbourne/OpenGD77/issues/688
Your hotspot is stuck in the broadcast because HOTSPOT mode on the radio is over.
It also happened to me, but i also saw that the radio went into normal mode in the same time.