rogerclarkmelbourne/OpenGD77 #686: custom boot tune causes boot loop on RD-5R

gigawatts, 2020-07-20:
I was playing around with some custom boot tunes, and the ran into a big problem.

Setting the following string "worked", but the 50 value tones just made some speaker crackling sounds:
`38,2,0,2,38,2,0,6,50,2,0,2,50,2`

However, the issue came about when I set the following string:
`38,2,0,2,38,2,0,6,48,2,0,2,48,2`

This causes the radio to crash and reboot in an endless cycle. I tried re-flashing the firmware (thankfully holding SK1+SK2 while powering up still lands in FW update mode), but that must not overwrite the boot tone values. Holding SK2 while powering on does not reset settings.

In an attempt to re-load the stock RD-5R v2.1.6 sgl firmware with the Open(xxx) Firmware Loader, I get the error "Error. The firmware doesn't match the transceiver model.". I was, however, able to load the stock FW with the stock "update.exe", that allowed the radio to boot again, though obviously in the stock firmware. Re-flashing your Open firmware again, however, brings back the boot loop. I found [this section in the guide](https://github.com/rogerclarkmelbourne/OpenGD77/blob/master/docs/OpenGD77_User_Guide.md#resetting-the-settings) that says holding the orange button while turning the radio on will clear a custom boot melody, but that doesn't seem to work.

So I guess a few questions:
1) How can I clear those bad configuration values so I can use the open firmware again?
2) What are the allowed values for boot tones? Why did 50 "sorta work", but 48 prevents it from booting entirely?

The only reference to boot tone parameters I could find was [this thread](http://www.opengd77.com/viewtopic.php?t=490)

Update: I was finally able to reset the boot tone, by mashing combinations of buttons while it boot looped every ~1 second, but I couldn't tell you which did the trick (I think I might have been holding both the VFO and CALL key at the same time, but I am not 100% confident that is what did it).

rogerclarkmelbourne, 2020-07-20:
Possibly the "Orange" button on the RD-5R is the VFO/MR button.

You'd need to check the code.

I've no idea why those high pitch sounds don't work.

I don't have time to fix this.

Please submit a PR to fix it.

gigawatts, 2020-07-20:
Understood. I'll have to test some more values, see what it will and won't take, and dig into the code to see how much of it I can understand.

In the mean time, if I want to make some additions to the user guide, is it fine to just make PRs to the .md copy of the guide? I assume you have a script to convert the MD version to the HTML and PDF copies?

rogerclarkmelbourne, 2020-07-20:
There is a nodeJs plugin to which does the MD to PDF and seems to use HTML as an interim format.

rogerclarkmelbourne, 2020-07-22:
Note number was not being bounds checked. The note number of 48 was beyond the end of the note frequency table, which resulted in an invalid note freq.

Extra note frequencies have been added and the note number is now bounds checked.

closed 2020-07-22 by rogerclarkmelbourne

gigawatts, 2020-07-22:
Cool! What are the valid ranges for notes? If you haven't already, I'd like to PR that into the guide.

rogerclarkmelbourne, 2020-07-22:
1 - 63

0 means no note and is used to signal the end of the data i.e 0,0 is used when people don't want a boot melody.