rogerclarkmelbourne/OpenGD77 #688: Hotspot exiting when TX power >250mW

mti-sk, 2020-07-22:
Hi,
Transmission power can be selected for HOTSPOT mode, but this must be done before entering the mode. 
When using the transmitter in standard mode, the power is set with the Fn2 + left or right arrow keys. This setting is then transferred to the HOTSPOT mode as well. I have no idea if the radio transmits in this performance as well, but let's assume that it does.
Unfortunately, the problem is that if the power is set to a value greater than 250mW, then the HOTSPOT mode is terminated after a short while.
I've read several issues, for example: 
 #497 and #672,
 which apparently point to a similar problem, but haven't been described accurately enough.

Termination is by MMDVMHost:
**E: 2020-07-22 07: 26: 54.529 Unable to read the firmware version after six attempts**
Strangely, when the power is set to 250mW, this does not happen! Never !

Radio: Baofeng DM-1801
Firmware: 1aeb6efb Jul 19 2020
Pi-Star: 4.1.2
Modem Mode: OpenGD77:v0.1.4

1. Log: HOTSPOT TX pwr 5w - Exited:
I: 2020-07-22 07:26:20.240 MMDVM protocol version: 1, description: OpenGD77_HS v0.1.4 GitID #1aeb6efb (Radio:DM-1801, Mode:MMDVM)
I: 2020-07-22 07:26:20.261 Display Parameters
I: 2020-07-22 07:26:20.261 Type: OLED
I: 2020-07-22 07:26:20.609 DMR Network Parameters
I: 2020-07-22 07:26:20.609 Address: 80.250.21.206
I: 2020-07-22 07:26:20.609 Port: 62031
I: 2020-07-22 07:26:20.609 Local: random
I: 2020-07-22 07:26:20.609 Jitter: 360ms
I: 2020-07-22 07:26:20.609 Slot 1: disabled
I: 2020-07-22 07:26:20.609 Slot 2: enabled
I: 2020-07-22 07:26:20.609 Mode Hang: 20s
I: 2020-07-22 07:26:20.609 Info Parameters
I: 2020-07-22 07:26:20.609 Callsign: OM2MO
I: 2020-07-22 07:26:20.609 RX Frequency: 430725000Hz
I: 2020-07-22 07:26:20.609 TX Frequency: 430725000Hz
I: 2020-07-22 07:26:20.610 Power: 1W
I: 2020-07-22 07:26:20.610 Latitude: xx.809547deg N
I: 2020-07-22 07:26:20.610 Longitude: xx.166225deg E
I: 2020-07-22 07:26:20.610 Height: 0m
I: 2020-07-22 07:26:20.610 Location: "TOWN"
I: 2020-07-22 07:26:20.610 Description: "Slovakia"
I: 2020-07-22 07:26:20.610 URL: "http://www.mw0mwz.co.uk/pi-star/"
M: 2020-07-22 07:26:20.610 DMR, Opening DMR Network
I: 2020-07-22 07:26:20.610 RSSI
I: 2020-07-22 07:26:20.610 Mapping File: /usr/local/etc/RSSI.dat
I: 2020-07-22 07:26:20.617 Loaded 14 RSSI data mapping points from /usr/local/etc/RSSI.dat
I: 2020-07-22 07:26:20.617 DMR Id Lookups
I: 2020-07-22 07:26:20.617 File: /usr/local/etc/DMRIds.dat
I: 2020-07-22 07:26:20.617 Reload: 24 hours
I: 2020-07-22 07:26:21.610 Loaded 167787 IDs to lookup table - /usr/local/etc/DMRIds.dat
I: 2020-07-22 07:26:21.610 DMR RF Parameters
I: 2020-07-22 07:26:21.610 Id: 231202201
I: 2020-07-22 07:26:21.610 Started the DMR Id lookup reload thread
I: 2020-07-22 07:26:21.610 Color Code: 1
I: 2020-07-22 07:26:21.610 Self Only: no
I: 2020-07-22 07:26:21.610 Embedded LC Only: no
I: 2020-07-22 07:26:21.610 Dump Talker Alias Data: yes
I: 2020-07-22 07:26:21.610 Prefixes: 0
I: 2020-07-22 07:26:21.610 Call Hang: 3s
I: 2020-07-22 07:26:21.610 TX Hang: 4s
I: 2020-07-22 07:26:21.610 Mode Hang: 20s
I: 2020-07-22 07:26:21.610 OVCM: off
I: 2020-07-22 07:26:21.610 DMR Roaming Beacons Type: off
I: 2020-07-22 07:26:21.722 Interfaces Info
I: 2020-07-22 07:26:21.722 IPv4: lo:127.0.0.1
I: 2020-07-22 07:26:21.722 IPv4: eth0:192.168.xxx.xxx
I: 2020-07-22 07:26:21.722 Default interface is : eth0
I: 2020-07-22 07:26:21.722 IP to show: eth0:192.168.xxx.xxx
M: 2020-07-22 07:26:21.722 MMDVMHost-20200615_Pi-Star_v4 is running
M: 2020-07-22 07:26:31.670 DMR, Logged into the master successfully
M: 2020-07-22 07:26:37.696 DMR Slot 2, received network voice header from 9A2Y to TG 91
M: 2020-07-22 07:26:38.177 DMR Talker Alias (Data Format 1, Received 6/20 char): '9A2Y D'
M: 2020-07-22 07:26:38.177 DMR Slot 2, Embedded Talker Alias Header
M: 2020-07-22 07:26:38.177 0000: 04 00 68 39 41 32 59 20 44 *..h9A2Y D*
M: 2020-07-22 07:26:38.890 DMR Talker Alias (Data Format 1, Received 13/20 char): '9A2Y DMR ID: '
M: 2020-07-22 07:26:38.890 DMR Slot 2, Embedded Talker Alias Block 1
M: 2020-07-22 07:26:38.890 0000: 05 00 4D 52 20 49 44 3A 20 *..MR ID: *
M: 2020-07-22 07:26:39.602 DMR Talker Alias (Data Format 1, Received 20/20 char): '9A2Y DMR ID: 2190222'
M: 2020-07-22 07:26:39.602 DMR Slot 2, Embedded Talker Alias Block 2
M: 2020-07-22 07:26:39.602 0000: 06 00 32 31 39 30 32 32 32 *..2190222*
E: 2020-07-22 07:26:39.710 No reply from the modem for some time, resetting it
M: 2020-07-22 07:26:39.710 Closing the MMDVM
M: 2020-07-22 07:26:41.713 Opening the MMDVM
**E: 2020-07-22 07:26:54.529 Unable to read the firmware version after six attempts**
M: 2020-07-22 07:26:59.530 Opening the MMDVM

f1rmb, 2020-07-22:
Hi,

   This is not a bug, it's a well known problem when the Pi get too much RF back from the USB connection, hence USB bus is resetted by the kernel driver, with the result of the USB device change and MMDVMHost still use the configured one.
Check the forum about this how to try to fix this (ferrite beads and so on).
About the Hotspot's power setting, check the user guide, it's documented how to define power setting in mmdvmhost.ini file.

mti-sk, 2020-07-22:
Hi,
I don't understand that much. What does too much RF on USB mean?
Do you mean USB data signal interference when transmitting a radio signal?
We are not talking about USB current load problems here. Because I'm using RPi4 rev2 with a 5V / 4A source, where this should be solved.

**Is it possible that a USB HUB would help?**

Thank you so much for the answers.

mti-sk, 2020-07-22:
@f1rmb

Hi,
I understand that I should not write on already closed issues.
However, I do not know how to provide **important information** to the people who created the issue (in my opinion). It also happened to me that my issue was closed, and important information was in another one.
Regarding the information in the user manual, I would like to say that according to various discussion forums, it is obvious that the description is not sufficient or does not work (AFAIK).

Great job, I wish you a lot of ideas and perseverance.

f1rmb, 2020-07-22:
The link to the forum thread in the closed issue where you poste ( #672 ) https://github.com/rogerclarkmelbourne/OpenGD77/issues/672#issuecomment-656069302
This kind of hotspot _exiting_ is not related to the hotspot code, but to a hardware problem.
For further discussion, the related forum thread is the place to go, as almost everything is explained there. The RD-5R still has this problem, and we don't know if it's possible to fix it.
About the power setting, there is a table that gives the numerical value to put in the mmdvmhost ini  to have the correct power output setting. Just checked the user guide and something went wrong apparently as the table disappeared. It was:

Power | PiStar RFLevel
--- | ---
50mW  | 1
250mW  | 5
500mW  |10
750mW  |15
1W  |  20
2W  | 40
4W  | 80
5W  | 99
5W++ | N/A

rogerclarkmelbourne, 2020-07-22:
I'll add it back into the User Guide

I'm editing it now to add the changes to the TS and CC filtering

f1rmb, 2020-07-22:
Ah, okay, I just raised a PR for the table in html and md file ;-)

rogerclarkmelbourne, 2020-07-22:
User Guide has been updated with warnings about RFI when using hotpsot mode and information about the RD-5R having hardware issues which probably prevent it being used in hotspot mode

https://github.com/rogerclarkmelbourne/OpenGD77/commit/98b936e7747d9a8426861a4c2529b82c3c18e4c7#diff-866fed918dc133304446fb4d01749e68

https://github.com/rogerclarkmelbourne/OpenGD77/blob/development/docs/OpenGD77_User_Guide.md

closed 2020-07-22 by rogerclarkmelbourne