rogerclarkmelbourne/OpenGD77 #711: Hotspot mode may not be passing GPS TA data that it receives  to MMDVMHost

rogerclarkmelbourne, 2020-08-08:
There is a report on the forum that GPS data is not being passed through the Hotspot, to the network etc

I know that TA data is passed from the network and transmitted by the Hotspot, but I don't know if TA data received by the Hotspot is passed to the network, and if TA is passed to the network, whether the GPS data packets are passed to the network.

Unfortunately, I can't test where GPS TA is passed to the network, the only radio I have with a GPS is my RT-3S and it is not operational as its on my workbench, so that I can trace connections from the STM32 to the C6000 etc

mi-gri, 2020-08-08:
I'm almost certain that GPS data has nothing to do with the Talker Alias...
Both ist inband data, indeed. (Referring to https://www.etsi.org/deliver/etsi_ts/102300_102399/10236102/02.03.01_60/ts_10236102v020301p.pdf 5.4.2 and 5.4.3)

mi-gri, 2020-08-08:
I tried sending using my Hytera PD658 with Talker Alias enabled. In the MMDVM I don't see the Talker Alias forwarded from the OpenGD77 hotspot to the MMDVM.

rogerclarkmelbourne, 2020-08-09:
If that radio uses SMS for GPS then it won't work through the hotspot as the hotspot can only handle voice data packets, because th firmware as a whole can't handle SMS data, because there is insufficient documentation on the C6000 DMR chip to implement DMR data handling at the moment.

rogerclarkmelbourne, 2020-08-20:
Closed as this is not an issue, its just a non-supported feature

closed 2020-08-20 by rogerclarkmelbourne