rogerclarkmelbourne/OpenGD77 #719: Unverified bug report via email from 2e0ukh

rogerclarkmelbourne, 2020-08-17:
I am using the version from August 1st.stable.

 I've been enjoying using the search function where you setting upper and lower limits. And I was searching for DMR and it works brilliant.and all of a sudden I thought it had been a long time since I've heard anything digital. And I could not even pick up my hotspot I had to reset the radio.

 And secondly today this happened twice. I press the orange button on the top and I wanted to transfer the vfo to a memory. I gave the the channel a name. I press the green button to confirm. I found the channel in all channels. I am familiar with that feature from the HD1.
 The channel appears to have disappeared from all channels. It definitely was in there I double checked.

f1rmb, 2020-08-20:
Can't reproduce either.

rogerclarkmelbourne, 2020-08-20:
I think perhaps he was not saving the name change to the codeplug.

I did reply to his email, linking to the section in the docs about SK2+Green, but he didn't respond to my email :-(

rogerclarkmelbourne, 2020-09-26:
Closing as nothing from the OP for over 1 month

closed 2020-09-26 by rogerclarkmelbourne