rogerclarkmelbourne/OpenGD77 #721: Cancelling FM "Tone scan" leaves Rx set to the last scanned tone

rogerclarkmelbourne, 2020-08-19:
Cancelling FM "Tone scan" leaves Rx set to the last scanned tone, which is not ideal.

The CTCSS/DCS tone value prior to the scan should be saved and this value should be restored if the operator terminates the tone scan

rogerclarkmelbourne, 2020-08-20:
For example.

I'm listening to a repeater which needs CTCSS for Tx.

Rx is set to no CTCSS, and I start the Tone scan
After scanning, if the Tone Scan does not find a tone, I have to stop the tone scan.
I press Red to stop the Tone Scan (probably any key will stop the Tone Scan)
The Rx CTCSS is left at the last CTCSS tone which was scanned, and I am then not able to hear the repeater, and I must go into the VFO Channel Details, and set the CTCSS to None.

I think if the operator interrupts the tone scan, the CTCSS should be set back to the value before the Tone scan was started

f1rmb, 2020-08-20:
Hi Roger,

   Yes, I know the problem, but the other case, the user is scanning to find the correct code, and he wants to stop the scanning process, but keeping the found tone set. So, if scan is paused, and green button is pressed, tone is kept, in any other case, the previous (before scanning) tone is restored.

Is that sounds correct ?

rogerclarkmelbourne, 2020-08-20:
Doesn't the tone scan stop when it finds the correct tone.

It scans multiple tones per second, so I don't know how the operator could effectively stop the tone scan on a specific tone.

f1rmb, 2020-08-20:
I can't remember if it stops.

Of course it's not quite possible to top at a given tone. I was talking if it finds the correct tone. I need to check the code about the stop or pause status.

f1rmb, 2020-08-20:
Okay, just checked, and yes, it stops when something come out of the speaker.

rogerclarkmelbourne, 2020-08-20:
OK.

I think if we store the Rx CTCSS before the scan and restore it if the scan is interrupted, it should fix this problem

f1rmb, 2020-08-20:
Yes, I'm adding this to the open PR

rogerclarkmelbourne, 2020-08-20:
OK. Thanks

rogerclarkmelbourne, 2020-08-20:
Fixed by 991f97913642125e4c62da788894bb4a99395523

closed 2020-08-20 by rogerclarkmelbourne