rogerclarkmelbourne/OpenGD77 #765: Option values, units do not translate for s = seconds

rogerclarkmelbourne, 2020-09-25:
Option values that show the time in seconds as "%ds" are not correct for some languages, including Finnish

It also causes problems with the voice prompts for Chinese as well as for Finnish

Adding "s" to the language strings may be overkill just for Finnish, as the Second is an SI unit.

But the units probably need to be voiced separately from the values on some screens.

f1rmb, 2020-09-25:
Idea:

```
typedef enum
{
  VOICEUNIT_SECOND,
  VOICEUNIT_MINUTE
   ...
  VOICEUNIT_HERTZ
  VOICEUNIT_KILOHERTZ
   ...
} VoiceUnit_t;

voicePromptsAppendValueWithUnit(char *value, VoiceUnit_t unit)
{
   // Check a flag in the VP header that describes the order (value + unit or unit + value), and spell it.
  ...
}
```

rogerclarkmelbourne, 2020-09-25:
OK.
An enum for the units may stop some duplication of the code..

f1rmb, 2020-09-25:
And about the function itself:

```
void voicePromptsAppendValueWithUnit(char *value, VoiceUnit_t unit)
{
     bool unitFirst = voicePromptGetHeaderUnitFirst();

     for(int8_t i = (unitFirst ? 1 : 0); (unitFirst ? i>= 0 : i < 2); (unitFirst ? i-- : i++))
     {
	  switch (i)
	  {
	  	case 0: // Value
		     ... 
		     break;
	  	case 1: // Unit
		     ...
		     break;
	       
	  }
}
```

f1rmb, 2021-01-24:
Could we consider that one fixed (yeah, units are always at the end of the sequence, but all units are correctly spelled now).

rogerclarkmelbourne, 2021-01-24:
Yep.

I think most units are now spoken correctly.

closed 2021-01-24 by rogerclarkmelbourne