rogerclarkmelbourne/OpenGD77 #767: Chinese language support

rogerclarkmelbourne, 2020-09-25:
Its probably not a definite study, but I've tried to find the total number of unique Chinese characters used in the voice prompt file
and it appears to be this list.
(I've also included the numbers 0 to 9 and the decimal point.)

The total number is 247 characters, but I'm not sure whether we need Tx and Rx and also DMR and FM, because if so, Chinese language support may not be possible as the chars are not unicode.

Also I guess A to Z are needed for channel names

点0123456789d信道联系人 输入兆赫千通话组时隙频率模式秒nu分伏瓦毫百之接收发射井号星绿色红版本宽窄加减交换与拷贝到至功扫描过滤区域电源码兹中文菜单致谢z强度by池l回溯界面w固件息选项设定显示声音详情g语言新列表热编译于k键盘已锁按蓝开请解关机错误仅超出业余段呼用户方受vq静噪快捷哑所有跳转存储类型无保重复不适带步进器全部是否"<=>/警告恢厂置校准参数波限制量麦克风增益长间速亮最小对比反正常背光延迟的忽略持暂停空删除？使在择辑低自动手允许结束一行两顺序提者都控阈值别名二零景.

michae1eo, 2020-09-27:
I think if I squeeze some of the words and substitute common and universally understand word (Rx,Tx,DMR,etc.) It can be 256 chars or less. 
When doing the translation of the user guide, I actrually tried to translate the english.h file to chinese.h, though technically it's not a language file. Since the .h file already consists all of the phrases exists in the UI, I can then make sure the translation of one word is always consistant throughout the entire user guide.
I attached the "Chinese.h" so maybe you can write some script and find unique chars. It should be more accruate.
[Chinese.txt](https://github.com/rogerclarkmelbourne/OpenGD77/files/5288602/Chinese.txt)

FlyGoat, 2020-10-25:
I'm trying to work on Chinese support, but there are some problems.

Firstly our print system only allows a byte for per character, and obvious not enough for Chinese. My solution is to use a free slot in our charset to indicate next byte would be a Chinese character.

Secondly I can't find suitable Chinese bitmap fonts. Actually Chinese bitmap fonts usually needs 16 points height to be recognizable. I'm using a 7px font to experiment. That's the smallest font I can find, but still has a extra row compares to FONT_SIZE_1.

rogerclarkmelbourne, 2020-10-25:
OK.

Most items in the first row are number or acronyms e.g "DMR" "FM" "W" "%" "C1"

You could potentially replace them with icons, but I think mostly Chinese operators may need to use use the European texts for these, as it would be impractical to display Chinese characters this small, especially as some texts are displayed in **Bold** to convey additional informtion

dan91ng, 2020-12-19:
translate only important functions (for example menus and Channel name) into Chinese

f1rmb, 2020-12-19:
That won't change the problem, which is the fonts used to display the texts.

f1rmb, 2021-01-24:
I guess that one could be closed, as it was stated, on the forum, that Chinese support is almost impossible, due to font system limitation, number of required glyphs, etc...

rogerclarkmelbourne, 2021-01-24:
I'll leave it open, as I'm sure we will be asked about this again if I close it.

But I've marked it as "Can't fix"

f1rmb, 2021-01-24:
Okay ;-)

Arnie97, 2021-02-09:
@michae1eo Thanks for your `Chinese.h` translation; I'll base my localization work on it.
I'll definitely want custom Chinese channel names though, so picking a limited set of Chinese glyphs would be a no-go. 

@FlyGoat IMO `FONT_SIZE_1` won't cause a severe problem -- it's 6x8 rather than 6x6. Chinese translation texts tend to be significantly shorter than Indo-European languages, so even if each single glyph occupies more territories on the screen (and more bytes in the ROM), the total size for a text string could be less. I'm working on a `PLATFORM_RD5R` fork that modified `ucPrintCore()` to share the same 8x8 Chinese font between `FONT_SIZE_1`, `FONT_SIZE_2` and `FONT_SIZE_3`.  As for `FONT_SIZE_1_BOLD`, it seems to be used for the `FM`, `DMR` and other hard-coded status indicators only, so I guess that won't be a problem neither, and I'll just keep it untouched.

Readability would be a pita though. I own a personal license for the [Dinkle Bitmap Font](https://3type.cn/fonts/dinkie_bitmap/index.html) so I'm using it in my current experiments. I'm unsatisfactory with the readability of the 8x8 font (including 1px padding between glyphs), let alone 7x7. Although 10x10 or 12x12 Chinese fonts would be pretty recognizable, choosing them would involve intrusive patches (changes to most UI layouts, if not all) to the codebase and make my fork difficult to be merged with upstream updates, so I'm still hesitating.

73 :-)

rogerclarkmelbourne, 2021-02-09:
Guys, 

Unless you can reduce the total number of glyphs to 256 then this is never going to work, as the entire firmware only supports 8 bit character codes.

xcvista, 2021-03-09:
I wonder if the chip can support bigger Flash chips? If so, maybe we can introduce UTF-8 first for best possible future compatibility, then just whack a subset of WenQuanYi font in here?

f1rmb, 2021-03-09:
Encoding is not a problem, is the storage, as explained multiple-tine. Fonts are also part of the firnware. The current system supports 256 different character, and Latin ones are mandatory. So, only a few are remaining free, even if the other non-latin are stripped out.
We added Katakana, as the required number glyphs is small, but it's already a separate build. 
Also, UTF-8 is one to four byte per char, and that's a lot of space.

closed 2021-04-20 by rogerclarkmelbourne