rogerclarkmelbourne/OpenGD77 #770: No way to set squelch to a non-custom squelch level

rogerclarkmelbourne, 2020-09-26:
If the internal squelch value is 0 (zero) on a Channel of the VFO, then the default squelch for that band (as defined in the Options) is used instead of the custom squelch.

However, AFIK at the moment once the user changes the squelch in the VFO (or potentially in a Channel), there is no way to return the squelch back to the default for the band.

I think perhaps the only way to implement a UI to change this to a non-custom squelch, would be to add a option to the Channel Details screen, either as a way of setting both the custom squelch value and changing to the squelch for the band, or possibly simply a way to switch between custom and band squelch.

f1rmb, 2020-10-20:
Maybe KEY_LEFT && KEY_RIGHT could be used as a reset ?

rogerclarkmelbourne, 2020-10-20:
We could just add another line in the Channel Details screen, below the Ch Power, because the functionality of the Channel squelch is the same as the Channel power.
I think this would be the most consistent thing to do.

f1rmb, 2021-01-24:
Hi Roger,

  I think this one can be closed.

Cheers.
---
Daniel

f1rmb, 2021-03-06:
Ping ;-)

closed 2021-04-20 by rogerclarkmelbourne