rogerclarkmelbourne/OpenGD77 #781: (Bug posted on the forum)  Voice prompt keeps repeating channel name when a signal is found while scanning

rogerclarkmelbourne, 2020-10-02:
https://www.opengd77.com/viewtopic.php?f=11&t=1288

Set the voice prompt level to 3. When a signal is found while zone scan, the channel name is repeated, hence the signal is not heared.

Observed on GD77 with R2020.10.01

f1rmb, 2020-10-02:
I think that was introduced with vk7js patches, at updateScreen() has been
used to respell things (iirc with a new function argument).

Will check later today or tomorrow morning.

---
73 de Daniel, F1RMB
http://f1rmb.blog.free.fr

rogerclarkmelbourne, 2020-10-02:
OK

Probably best if we eventually revert all those changes, it looks like they caused a lot of damage

f1rmb, 2020-10-02:
I'm afraid it is, looks have a strong side effect, I will closer then tell
you.

Cheers.
...
Daniel

rogerclarkmelbourne, 2020-10-02:
OK

rogerclarkmelbourne, 2020-10-04:
Fixed by Daniel

closed 2020-10-04 by rogerclarkmelbourne