rogerclarkmelbourne/OpenGD77 #787: Channel name seems to be announced on Voice level 3 when a DMR signal ends

rogerclarkmelbourne, 2020-10-04:
Channel name seems to be announced on Voice level 3 when a DMR signal ends

f1rmb, 2020-10-04:
Hi Roger,

Humm, while scanning ? Because I just checked on normal operation, and I hear nothing.

Cheers.
...
Daniel

rogerclarkmelbourne, 2020-10-04:
No. Not during scanning.

I think it announced just when a signal ended

f1rmb, 2020-10-04:
Hi Roger,

I just tried while scaning, I don't noticed such problem.
Do you have a procedure to reproduce ?

Cheers.
...
Daniel

f1rmb, 2020-10-04:
Hmmm, Ok.

f1rmb, 2020-10-04:
I can only see a problem with loadChannelData(), being called when start RXing.

rogerclarkmelbourne, 2020-10-04:
OK.

I will need to test, because I'm not sure under what conditions it occurred.

I just noticed that the radio was not scanning, and suddenly it announced the channel.

It may have rebooted, as I see that if I turn the radio off and on the channel is announced.

f1rmb, 2020-10-04:
Yes, the channel name is announced while booting on L3

rogerclarkmelbourne, 2020-10-04:
ummm

I can't replicate the bug again.

I will close the issue, because perhaps the problem is a random reboot.

closed 2020-10-04 by rogerclarkmelbourne