rogerclarkmelbourne/OpenGD77 #793: DTMF Settings Not Working Correctly

g7jiq, 2020-10-04:
This is NOT a complaint, just an observation.

The DTMF settings in latest software do not appear to change the tone speed and the numbers from the radio contact list are sending to slow and without gaps.
Because of this, an allstar nodes ignore the DTMF commands sent.
Also in the latest software, the signaling system in the dropdown menu appears to do nothing.

g7jiq, 2020-10-04:
An update, 
I have updated the radio to the latest (unstable) ver dated today.
The DTMF is now changable within the software, 
And the radio dialer functions fine,
There is one slight problem, 
The radio drops tx to soon after the last tone, unfortunatly the allstar node needs a slight carrier after the last tone finishes for it to decode the DTMF string.

Is there anyway to have the hang time adjustable ?

f1rmb, 2020-10-04:
I will add a the last silent duration at the end of the sequence.

Thanks for the report.

f1rmb, 2020-10-04:
The duration of the tail will be equal to the "First Digit Delay" value.

g7jiq, 2020-10-04:
That would be great, I await an update file

g7jiq, 2020-10-05:
I have updated both the radio and software, 
I get the tx start delay (be nice if it could be set longer, set at 1000) 
But no tx hang after the tones have sent, it almost seems like the last tone is cut off while being sent.

rogerclarkmelbourne, 2020-10-05:
Bug fix is in progress, and will be released later.

Changes to longer delays will require change to the CPS, but I think this should be independent of the firmware, as the firmware does not check the value from the CPS.

rogerclarkmelbourne, 2020-10-05:
End delay is now implemented in the latest development release files

g7jiq, 2020-10-05:
That has fixed it, 
It is working just as required, and the dialler list is spot on,
Thankyou very much ...

rogerclarkmelbourne, 2020-10-05:
@g7jiq 

No worries

BTW. Can you give me a use case for why the initial and hence the tail delay would need to be longer than 1000mS

I think I could change the CPS to allow it, but I don't add features which don't have a specific use case, as it just complicates things who don't fully understand what values these settings should be.

Also.

I think there are several "Reserved" data fields in the codeplug data structure, (which was inherited from the official cps and firmware)

Potentially the tail delay could be put into one of those "reserved" locations, because from past experience those fields are usually set to zero, which can be then used as a flag to tell the firmware whether a specific value has been configured.

This does rule out tail values of 0mS, but I don't think that's a big problem

g7jiq, 2020-10-05:
The inital delay over 1000 was to let the rx radio on my allstar node wake up,  I thought this was a quirk of the radio, (and was like this since built) but it turned out to be the rx going to sleep in power save mode, I have since remove this. 

The hang time was the main problem I found, but once enabled the last digit was abled to be read correctly, (with a instant drop I think the squelch tail noise from the rx radio corrupted the last dtmf tone, just an idea but possible)

I found these settings to work for me.
First Digit Delay  800
First Digit Duration 100
*# Duration 100
DTMF Rate 6

rogerclarkmelbourne, 2020-10-05:
OK.

Some

g7jiq, 2020-10-05:
Yes I read that, 
I downloaded and installed the OpenGD77.sgl file.
Does the unstable 2020.10.04 downloader in the CPS install the same file?

rogerclarkmelbourne, 2020-10-05:
CPS gets the latest unstable from Github.

The last version I uploaded was around 12 hours ago

I think the git revision code is 55482d8 (see firmware info).

(This is my local git revision code, it does not relate to github)

rogerclarkmelbourne, 2020-10-05:
BTW.
I don't make new releases for every single change, it would be far to difficult and also extremely confusing.

I just open a new pre-release  and keep updating it with bug fixes, until it becomes stable, usually in 1 to 2 months, and then that version becomes the new stable release

I don't commit unstable code to github, as it causes all sorts of problem, because people build and distribute unstable copies via FB etc, and I think get loads of reports of bugs on what people think is the stable release

marrie, 2020-10-05:
Running the latest unstable firmware, I still cannot get the dtmf tones to work, see the last log in the issue I submitted about this. I can if needed make a tiny recording of what happens, or not. I did hear on a repeater   I was on when I tried a disconnect command all but the last tone.  I knew it didn't work when I heard my tone fire, and not the sound indicating I disconnected. it did work when I did disconnect by diling the tones by hand.

f1rmb, 2020-10-06:
@marrie did you ever tried to change the timing/rate settings ? See above what @g7jiq wrote.

marrie, 2020-10-06:
Thanks. I did, but I still need to enter a dummy digit at the end.  here are my settings in the DTMF section.

I have 500 for the first field,  100 for the second field, 200 for the third and 5 for the rate. There is still a bug somewhere as i should not have to enter the dummy digit at the end. Some can get it working with out said digit.

f1rmb, 2020-10-06:
You also have ton of settings in asterisk.

Try to increase the fist digit delay (it increases the tail symmetrically).

marrie, 2020-10-06:
I'm actually not as tech savvy in asterisks. What conf file do I edit to 
  do this. Once I started the asterisks server on my node everything was 
grate, but I still need to enter the dummy tone. It does not catch that 
according to the log I see. Would editing those settings help? I think a 
bit of documentation under advanced will be needed for those of us who 
want to use this. Not good at writing this but am taking some notes as I 
go.

rogerclarkmelbourne, 2020-10-06:
Perhaps we need to add some sort of default tx tail.

Or perhaps let the CPS specify the tail length and default it to 100mS

f1rmb, 2020-10-06:
@marrie I don't know, I just quickly searched about asterisk and DTMF.
Did you tried various values with First Digit Delay ?

marrie, 2020-10-06:
I did. No matter what I do, I need that last dummy tone. For example *76 has to become *766 or asterisks won't see the 6 at the end of the string. I've seen this when reading, in real time, the client. and for some reason the only time it does, or did work as of yesterday is when I started the client on my windows machine via SSH.and I could see the log entries. It's odd, and does need to be looked into, but this works for now.

Can there maybe be an end or tail that can be defaulted to 100ms but one can if necessary raise it to something not more than 1000 ms? ? That is long enough IMHO.

marrie, 2020-10-06:
# update

A fellow ham discovered that if you key up once after cold starting the radio, the dummy tone is not needed. this I know is a problem with one of the softwares, but I'm just letting you devs know.

I've just confirmed this, and it works. So, in saying all of that, can that be taken a look at, as that should not be necessary, unless this is by design.

g7jiq, 2020-10-06:
I have updated with the latest OpenGD77.sgl file, 
This sorted my radio 100%, with a nice hang after the last DTMF tone, (settable via the cps)

The hang time was the main problem I found, but once enabled the last digit was abled to be read correctly.  (with a instant drop I think the squelch tail noise from the RX radio corrupted the last dtmf tone, just an idea but possible)

I found I had to key up first to let the RX radio on my allstar node wake up,(after it had been quiet for a while) I thought this was a quirk of the radio, (and was like this since built so I didn't pay much attention to it) but it turned out to be the RX going to sleep in power save mode, 
I have since remove this and solved another problem I didn't know I had.

I found these settings to work for me on an allstar node using a baofeng BF-888S.

First Digit Delay 800
First Digit Duration 100
*# Duration 100
DTMF Rate 6

Hope this is of help..

marrie, 2020-10-06:
@g7jiq, thanks so much. I'm also using the unstable build, so for me the key up is necessary. I did not update to any unstable version of the CPS software as of yet. I'll just wait until that is released to the public, as I don't want to brake anymore than what is broken. \*smiles.\*

Again, thanks so much. I actually have the first tone at 100 and the star at 200, then the tone rate at 5 as my node can't really take anything faster than that.

g7jiq, 2020-10-06:
Something else I found when I first built it, was the hotspot is a bit touchy about the volume of the rx audio fed from the radio to the usb audio fob and the allstar setting for it to reliably decode DTMF.
It can be a bit of a fiddle to get it right, but once done it works a treat, and well worth it.

As I mentioned I am running the latest cutting edge file, installed by selecting the file not by auto installing via the download buttons.
So far I havn't found any problems or bugs using FM or DMR. 
Hope this is of help.

rogerclarkmelbourne, 2020-10-06:
I think the best way forward is for us to add a separate control for the tail duration.
There is an unused data field in the codeplug that could be used for this, with perhaps the default minimum of 100mS 

Also the Rate control does not have enough settings, so I think we should change it so that rates from 1 to 10 tones per second can be selected.

This will means that after the change, everyone using this version will need to update the codeplug, because the rate will be wrong (it will be lower), but its better in the long term to add this flexibility now, before loads of people are using it

marrie, 2020-10-07:
I'm up for  trying that. I like the 1-10 option as well for the nodes 
that cannot do faster  dtmf tones.
Thanks.

rogerclarkmelbourne, 2020-10-07:
OK.

rogerclarkmelbourne, 2020-10-07:
@marrie @g7jiq 

Daniel has added support for the DTMF tail, and also changed the DTMF rate selection so that rates from 1 to 10 can now be selected

Please download and install the latest CPS, 

https://github.com/rogerclarkmelbourne/OpenGD77CPS/releases/tag/R2020.10.07 

And also update your firmware from the development/unstable release., which has been updated with support for these changes

marrie, 2020-10-07:
Got it, and installed. I do not how ever see the latest release under the dev branch. I looked under releases  as well and the latest is from the forth of October in the ongoing dev section. If that is the one you mean, I'm running it now.

I also only still see under dtmf settings a first digit delay, and not a tail setting as well. this is, however after only a few minutes of poking around. I've attached a screen cap of what the window looks like.

Thanks.

[image]

f1rmb, 2020-10-07:
Yes, that's the one. Dev releases are rolling one, the date never change, you will see a timestamp with the changes since the last update.
What you're looking for is "DTMF Tail [ms}" in the "DTMF Settings" panel.

g7jiq, 2020-10-07:
Nice tweeks, all working as before.

rogerclarkmelbourne, 2020-10-10:
Thanks. 
I'll close this issue as the problem has now been resolved

closed 2020-10-10 by rogerclarkmelbourne