rogerclarkmelbourne/OpenGD77 #795: Power control no longer works on the GD77S since the update to the Channel power system

rogerclarkmelbourne, 2020-10-05:
GD77S code needs to be updated to use increasepower() and decreasepower()

rogerclarkmelbourne, 2020-10-05:
This should now be fixed in the latest Development / Unstable firmware files

closed 2020-10-05 by rogerclarkmelbourne

cjeffcoat2, 2020-10-05:
OK, I realize this is you is closed, I went into modify some individual channel power settings and noticed that the power level is set to master, I modified this to what I would like for individual channels but now all of the rest of my channels have incorrect power levels, where is the master power level set?￼

rogerclarkmelbourne, 2020-10-05:
It should be fixed 5 mins ago

rogerclarkmelbourne, 2020-10-05:
Individual power settings on the GD77S is not currently supported

cjeffcoat2, 2020-10-05:
OK, thank you so much I really appreciate all of the work that you are doing! It is fantastic and greatly appreciated.

rogerclarkmelbourne, 2020-10-05:
You can add that as an enhancement issue, along with the other enhancement issues (lots of them)

cjeffcoat2, 2020-10-05:
OK, sorry, posted this in the wrong spot it is for the 77.