rogerclarkmelbourne/OpenGD77 #802: Last digit is not entered, but sounded, when a DTMF  script is executed.

marrie, 2020-10-05:
I'm using the voice on the firmware, but hopefully the screen will say the same thing. Here are the steps to reproduce.

One. Hit the second key below push the top key and pound key together.
￼￼ two. Up and down arrow to the contact or the DTMF contact you want to execute.
￼ three. Hit the menu key, and then the menu key again.

￼ you will hear the tones sound, but when you look at the asterisks log in my case on a ClearNode, the last tone will not be executed. So for example if you have *76 as your script, only *seven will be at entered. I am on an analog allstar node testing this.￼

marrie, 2020-10-05:
We found a work round for now. Add an extra digit to the end of the string. What needs to happen is, if possible, increase the transmit time to allow for that last digit to actually make it through to the node. This way adding a dummy digit to the end of the script is not necessary.

f1rmb, 2020-10-05:
Hi,
   If I get it right, your problem is the system that receive the DTMF sequence, right ?
Did you tried the development version Roger released today, as a tail has been added at the end of the DTMF sequence:

[Link to the forum post](http://www.opengd77.com/viewtopic.php?f=13&t=1059&p=11419&sid=013a1de8aeaa6064ef8dd485a3ea55ef#p11419)

Cheers.
...
Daniel

marrie, 2020-10-05:
I have not yet tried, but here is the log from today. this is running the oct 4 build. Nothing actually connected, and I got a registration failure.

[Oct  5 11:51:29] NOTICE[376] chan_simpleusb.c: Got DTMF char *
[Oct  5 11:51:29] NOTICE[376] chan_simpleusb.c: Got DTMF char 3
[Oct  5 11:51:29] NOTICE[376] chan_simpleusb.c: Got DTMF char 5
[Oct  5 11:51:29] NOTICE[376] chan_simpleusb.c: Got DTMF char 1
[Oct  5 11:51:29] NOTICE[376] chan_simpleusb.c: Got DTMF char 4
[Oct  5 11:51:29] NOTICE[376] chan_simpleusb.c: Got DTMF char 4
[Oct  5 11:51:29] NOTICE[376] chan_simpleusb.c: Got DTMF char 8
[Oct  5 11:51:53] WARNING[365] chan_iax2.c: Received unsolicited registry ack from '216.244.83.154'
[Oct  5 11:51:53] WARNING[365] chan_iax2.c: Registration failure
[Oct  5 11:55:34] NOTICE[369] chan_iax2.c: Registered IAX2 to '44.72.21.13', who sees us as 75.163.21.136:4569 with no messages waiting

marrie, 2020-10-05:
Hi there, just updated to the beta version if you will, and am making progress. it still broke, but here is the entry now. It still looks like it is failing as I didn't even hear the tones through iaxrpt

# log follows
[Oct  5 12:04:01] NOTICE[376] chan_simpleusb.c: Got DTMF char *
[Oct  5 12:04:02] NOTICE[376] chan_simpleusb.c: Got DTMF char 3
[Oct  5 12:04:03] NOTICE[376] chan_simpleusb.c: Got DTMF char 5
[Oct  5 12:04:04] NOTICE[376] chan_simpleusb.c: Got DTMF char 1
[Oct  5 12:04:04] NOTICE[376] chan_simpleusb.c: Got DTMF char 4
[Oct  5 12:04:05] NOTICE[376] chan_simpleusb.c: Got DTMF char 4
[Oct  5 12:04:06] NOTICE[376] chan_simpleusb.c: Got DTMF char 8
[Oct  5 12:04:06] WARNING[376] channel.c: Unable to handle indication 13 for 'IAX2/45.56.107.100:4569-9988'

I hope that helps.

f1rmb, 2020-10-05:
But what is the full DTMF sequence ?

marrie, 2020-10-05:
the full sequence is

*350448 for example

I also cannot get 

*76

to work either.

Here is my log for both, first, then I used the dtmf *76 then when I keyed it up on my own.

[Oct  5 13:13:13] NOTICE[376] chan_simpleusb.c: Got DTMF char *
[Oct  5 13:13:13] NOTICE[376] chan_simpleusb.c: Got DTMF char 7
[Oct  5 13:13:17] NOTICE[370] chan_iax2.c: Registered IAX2 to '216.244.83.154', who sees us as 75.163.21.136:4569 with no messages waiting

and now when I tnered it on my radio's key pad.

[Oct  5 13:13:32] NOTICE[376] chan_simpleusb.c: Got DTMF char *
[Oct  5 13:13:33] NOTICE[376] chan_simpleusb.c: Got DTMF char 7
[Oct  5 13:13:34] NOTICE[376] chan_simpleusb.c: Got DTMF char 6

marrie, 2020-10-05:
Looking a the log a bit further, It dropped the 6 out of the *76. I'm running the letr's say unstable beta build of the firmware. Should this work on all nodes such as my  clear node? I'm thinking yes? I might put in a suggestion to hear how your dtmf script will sound before putting it into the radio, so you can make the necessary adjustments in the dtmf settings to have your node(s) pick the tones up correctly.

Also, it seems like the * is being entered in way too long, and the others are being entered in a bit more quickly. Is this supposed to happen? the DTMF settings are not actually that clear in the case of the program.

rogerclarkmelbourne, 2020-10-05:
Increase the initial delay as this is also used as the tail delay.

see

https://github.com/rogerclarkmelbourne/OpenGD77/issues/793

marrie, 2020-10-05:
Nope, it still broke. Here is the entry when trying to enter *351448. I don't see any of my numbers in this time.

[Oct  5 14:46:40] WARNING[363] chan_iax2.c: Received unsolicited registry ack from '178.32.160.85'
[Oct  5 14:46:40] WARNING[363] chan_iax2.c: Registration failure
 but when I enter the same command from the radio itself, it works. Is there something I'm missing somewhere?

rogerclarkmelbourne, 2020-10-05:
This is a duplicate of issue https://github.com/rogerclarkmelbourne/OpenGD77/issues/793

I will close this issue, please use the issue #793

closed 2020-10-05 by rogerclarkmelbourne