rogerclarkmelbourne/OpenGD77 #808: Apply priorities to each channel when scanning

rogerclarkmelbourne, 2020-10-06:
See

https://www.opengd77.com/viewtopic.php?f=4&t=1159&p=11444#p11444

rogerclarkmelbourne, 2020-10-06:
Potentially the ScanList byte in the channel data struct could store this a scan priority

f1rmb, 2020-10-06:
Or one of the unused flag4 bit (e.g. AutoScan, LoneWorker, AllowTalkAround).

rogerclarkmelbourne, 2020-10-06:
Yes. It depends if just one channel is the priority or whether each channel can have its own priority number

vk7js, 2020-11-28:
It would be good if a specific priority, e.g. 0, would actually interrupt another channel if scanning has stopped on that channel. I'd be happy with just one priority channel which would pre-empt whatever channel or vfo was currently playing. Giving other priorities to channels though an interesting idea might be too complicated. In lieu of dual channel mode, a single priority channel is very desireable. Perhaps an option could decide if the priority channel pre-empted another active channel or if it merely was checked for activity while monitoring a 2nd channel.

VA3NDP, 2020-11-28:
I concur. The scan system is very well done and very fast, the addition of a priority type system would be extremely helpful.

closed 2021-04-20 by rogerclarkmelbourne