rogerclarkmelbourne/OpenGD77 #815: bug: sometimes the voice skips something when I arrow too fast

marrie, 2020-10-08:
# description

When I arrow up and down quickly, sometimes the voice skips what ever I passed over, and I need to either press down and up arrow, (same with left and right arrow) or use my key 1 to read what ever was skipped

# steps
after installing a voice into the radio via the software do the following:

1. turn on the radio
2. use it as normal making sure the prompt is set to 1. I did get this to happen in 2 and 3 as well.
2. arrow up and down very quickly.

Notice sometimes you will hear the voice try and speak, but  nothing happens.

# workaround

you can either down and up arrow  or use key 1 to read what was said. This is a random, hard to reproduce bug I noticed yesterday. I ntoiced the same thing can sometimes happen with the left and right arrow as well if you go way too fast and the thing happens to still be speaking.

rogerclarkmelbourne, 2020-10-10:
Thanks, this is a known bug, but I forgot to log it.

marrie, 2020-10-10:
Oh good, I'm not going crazy.

I hope what I wrote makes sense. I have not tested all of the voices as 
I don't speak any other languages. Hopefully someone else can test other 
languages and see if they also have this issue as well.

Thanks.

rogerclarkmelbourne, 2020-10-10:
No,

I've heard it myself, but as it isn't a serious bug, or a high priority enhancement, I forgot to log it.

marrie, 2020-10-10:
That's fine. Wanted to make the staff aware is all. Thanks so much for 
all you do.

Thanks.

rogerclarkmelbourne, 2020-10-10:
@marrie 

Just one small thing.

You used the word "staff" in your last post

This is **NOT** commercial software.  There are no **staff**  just volunteers.
Currently there are only 2 active volunteers on the firmware development.

The entire firmware was developed by only a handful of people, mostly by 3 people.

rogerclarkmelbourne, 2020-11-24:
FYI.

I've now tried to investigate this problem several more times but unfortunately I have not been able to track down the problem.

marrie, 2020-11-24:
Thanks. It is still happening with the build I'm running 094023NOV122020 
I don't have the git commit right now. I couldn't actually hear it 
clearly, my fault, not the radio. Thanks again.