rogerclarkmelbourne/OpenGD77 #827: Channel -> VFO does not set the correct TS

rogerclarkmelbourne, 2020-10-18:
When using the quick menu to copy the current Channel to the VFO, the current TS / override is not copied to the VFO.

This is confusing, because if you are listening to a station, when the channel is copied to the VFO, the station will not continued to be heard if the VFO was not set to the same TS as the Channel

PedroPauloBrazil, 2020-10-18:
Hello, Roger! I use a channel, with default TS = 2, a talkgroup list with several TG, all without a TS override, except one, which has TS override = 1. In firmwares prior to R20201016, TS only changed to TS = 1 when I navigated to the TG that has the override returning, to TS = 2, whenever I selected any other TG on the talkgrouplist. It happens, that in the version R20201016, once I select the TG that has the TS = 1 override, the channel default becomes TS = 1, even if I browse the other TGs that do not use the TS override. This appears to be a bug, which prevents the correct use of TS override. Could you please see what happens? Thank you very much and congratulations on the fantastic OpenGD77! 73 by Paulo - PU2OIP

rogerclarkmelbourne, 2020-10-18:
@PedroPauloBrazil 

You posted to an unrelated issue.

I have created a new issue for you and replied in that issue and I will remove your comment

See

https://github.com/rogerclarkmelbourne/OpenGD77/issues/828

rogerclarkmelbourne, 2020-10-30:
I think this has now been fixed by Daniel

closed 2020-10-30 by rogerclarkmelbourne