rogerclarkmelbourne/OpenGD77 #828: Confusion about TS override - posted to different issue

rogerclarkmelbourne, 2020-10-18:
https://github.com/rogerclarkmelbourne/OpenGD77/issues/827#issuecomment-711246638

Hello, Roger! I use a channel, with default TS = 2, a talkgroup list with several TG, all without a TS override, except one, which has TS override = 1. In firmwares prior to R20201016, TS only changed to TS = 1 when I navigated to the TG that has the override returning, to TS = 2, whenever I selected any other TG on the talkgrouplist. It happens, that in the version R20201016, once I select the TG that has the TS = 1 override, the channel default becomes TS = 1, even if I browse the other TGs that do not use the TS override. This appears to be a bug, which prevents the correct use of TS override. Could you please see what happens? Thank you very much and congratulations on the fantastic OpenGD77! 73 by Paulo - PU2OIP

rogerclarkmelbourne, 2020-10-18:
R20201016 fixed a problem where the TS override was constantly reset, and other bugs where TS priority order was incorrect.

This was not correct. TS override should not have been reset when TG was changed via Left or Right arrows.

Channel TS has lowest priority.
Contact TG with TS override has next highest priority
Operators selection of TS overrides the current TS, however if you change TG to a new TG with TS override, this will change the TS to match the TS required for the TG

If you are using the TS in the Channel then you probably didn't construct your codeplug correctly.

PedroPauloBrazil, 2020-10-18:
Ok, Roger, I understand that the priority of TS override is higher than that defined in the channel. And that's exactly what I use in OpenGD77!

The logic in my codeplugs for OpenGD77 consists of using several talkgroups in a TGlist, associating that TGlist to a channel and using that channel (along with navigation by the side arrows) to switch between these different talkgroups.

That's exactly the problem: I understand that the SCOPE of the TS override of any TG (defined in the contacts) must be limited or restricted to this TG (or contact) and that in the absence of the TS override in the other TGs of the TGlist, in navigation side, the TS must change to the TS defined in the channel. 
That, or else I can't understand the usefulness of TGlists, TS overrides, TS on channels, etc., just what I always thought was a wonderful concept in relation to other DMR radios.

Finally, I thank you for your kindness in answering me and correcting my post. I didn't want to create a new issue, thinking that the subjects were related, but I understand that this was the result of my inexperience on GitHub. I promise to learn! :)

With my best regards, Paulo - PU2OIP.

rogerclarkmelbourne, 2020-10-18:
@PedroPauloBrazil 

OK

Can you explain to me why the changes to the priority don't work for your usage?

Which network do you use BM or MARC or TGIF

etc

PedroPauloBrazil, 2020-10-19:
I can't explain why, but after finding the TS override, the slot becomes the one defined in the override, even if I navigate to a TG that doesn't have it. Changes between channels and zones also do not change the TS.

I use Brandmeister, TGIF and DMR+. In the previous version my codeplug worked perfectly (maybe using a bug, I don't believe it, but I don't know ...).

Reading your answer, it seems that TS override does not have a restricted scope to the contact in which it was defined. I built my codeplug based on this scope and understood that for any contact or talkgroup that does not have an override TS, the TS defined in the channel would be used, since there is no override defined in this case!

Yes, I know that the replacement can be canceled with a * long press, thanks! However, I think it would not even be necessary if the scope of TS override was restricted to the TG or contact in which it was defined. That is the question I have.

I can also send you my codeplug, if you're interested, so that you can see it better, I just don't know how to send it so it won't be public. Regards, Paulo.

PedroPauloBrazil, 2020-10-19:
Just in time:

"For BM, if the Contact TG has no TS override and the operator does not press * to override, the TS will be the Channel TS."
- Yes, while I am on the talkgroup that has the TS override, it persists correctly. The problem is that it persists even when I navigate to another talkgroup or channel or zone, unless a TS override has been individually set on all TGs, which is incorrect, I think.

"Perhaps in your case you require the Channel TS to be higher priority then the operator press of * to set the TS"
- No, I also believe that the priorities are right. The TS defined in the channel must be of lower priority than any TS overrides.

rogerclarkmelbourne, 2020-10-19:
OK.

I think the change required may be for when there is No TS override on TG, and No operator TS override.

Currently I think that under those conditions perhaps the TS on the Channel is not honoured.

I'll can't work on this now, but I will look this evening (its 11:54 morning local time for me now..)

PedroPauloBrazil, 2020-10-19:
"I think the change required may be for when there is No TS override on TG, and No operator TS override."
"Currently I think that under those conditions perhaps the TS on the Channel is not honoured."

Ok, Roger, I understand that this is exactly what you described above.

As for when you can make the changes, please rest assured! What I wanted to do was report what happens. Now, the case is in good hands. :)

It is already 10:10 pm, local time, here I am going to turn things off here in the shack and watch some tv. ;)

Have a good lunch and a good week!

Paulo

rogerclarkmelbourne, 2020-10-19:
It's Monday already in here in Melbourne ;-)

rogerclarkmelbourne, 2020-10-19:
Umm

I checked the code, and if there is no TS override on the TG, and also no manual TS override, then the TS should be set to the TS set by the Channel "Repeater Slot"

However, I have not tested this to confirm that this is functioning correctly

PedroPauloBrazil, 2020-10-19:
Good Morning! Yes, the logic seems impecable. Perhaps the tests show the existence of a small bug.
If there is anything I can do here, I will be available. 73!

rogerclarkmelbourne, 2020-10-21:
I am trying to reproduce the bug which you have

I have 
"Channel A" set to TS1
"Channel B" set to TS2

I have my contact is TG9 with  Repeater slot override Disabled
I did a long press on * to clear any TS override

If I change channel using the Up and Down, I see the TS changes depending on if it is Channel A or Channel B

If I press the * to change the TS,  e.g. to TS 1, when I change channels, the TS remains as TS1.
If I press * again to change the TS to TS2. when I change channel (from Channel A to Channel B, the TS remains on TS2

If I change TG on eithe Channel A, or Channel B, to a TG which a TS override e.g. TG 5 is TS1, and TG3803 is TG2.
Then the TS always changes to the TS of the TG. 

If on TG3803 TS2, I press * the TS changes to TS 1
If on TG5 TS1, I press the * the TS changes to TS2

If I long press on *, the override is removed and the TS changes to the TS for the TG.

I then assigned different TG Lists to Channel A and Channel B

Channel A  I assigned my Brandmeister TG List
Channel B I assigned my DMR MARC TG list

The first TG in my Bandmesiter TG List is the TG OpenGD77 on TS1
The first TG in my DMR MARC TG List is TG 505 TS2

When change from Channel A to Channel B , I see the TS changes from TS 1 on Channel A where the TG is OpenGD77 on TS1, to TS 2 on Channel be because the TG is 505 TS2

I then changed Channel A to TS2, and re-tested and for the OpenGD77 TG, the channel still changes to TS1, because of the TG of the TG.

Please can you explain in more detail, exactly the TS on each channel and the TS on each TG, and when the logic does not work or perhaps there is a bug.

tattoli, 2020-10-21:
I have also encountered this behaviour changing since I updated from the latest stable September release to the 10/16 release yesterday, with the time slot manually selected using * and using a contact with no time slot attached to it, the channel is set to ts1, but when I reselect the contact with no ts associated with it or hold the * key and it resets, all of my dmr channels get set to ts2.
In the previous firmware if the ts had been manually selected and then I selected a contact with no ts, the radio would default back to the channel ts but for some reason now its getting forced to ts2 on all of my channels. There does not appear to be a combination of settings that i can choose on the new firmware to even get the channel ts to show up correctly at all.

PedroPauloBrazil, 2020-10-21:
Okay, Roger, I'll try to describe my scenario.

1) Let's agree:

TG = Talkgroup
TS = Time slot
TGL = Talkgroup list
ChTS = Channel Time Slot
TSO = Time slot override (defined at contact level. Default = disabled)
MTSet = Manual TS set (alternates between 1 and 2 by pressing * key.)
ClrTS = Long-press of the key *

2) "Codeplug definition".

The codeplug used in the tests has two channels, A and B, both Brandmeister, for simplicity.

2.1) Channel A

ChTS-A = 2
TGL-A = 724, 724942, 724946, 724947, ...
The contact for the TG 724 has TSO = 1. All other contacts have TSO = disabled.

2.2) Channel B

ChTS-B = 2
TGL-B = 7241, 72411, 72412, 72413, 72414, 72415, ...
All contacts have TSO = disabled.

3) Tests

3.1) Turn on the GD77 and press ClrTS.

3.2) Navigation with Up and Down arrows

3.2.1) I'm on Channel A, TG = 724 and TS = 1 appears on the screen
3.2.2) I press Up and go to TG = 7241, TS = 2
3.2.3) I press Down and return to the conditions of 3.2.1
3.2.4) I press *. TG = 724 remains on the screen and TS changes to TS = 2. I press Up and the screen shows TG = 7241, TS = 2. I press Down and the screen returns to TG = 724, but TS REMAINS TS = 2. TS only returns to TS = 1 if I type ClrTS. In the previous firmware, if I remember correctly, it went back to TS = 1, considering the value of TSO = 1 of TG = 724.

3.3) Navigation with the Left and Right keys

3.3.1) I'm on Channel A, TG = 724 and TS = 1 appears on the screen
3.3.2) I press Right and go to TG = 724942, TS = 2
3.3.3) I press Left and go back to the conditions of 3.3.1
3.3.4) I press *. TG = 724 remains on the screen and TS changes to TS = 2. I press Right and TG = 724942, TS = 2 appears on the screen. I press Left and the screen returns to TG = 724, and TS BACK to TS = 1. Here, the operation seems to me identical to that of the previous firmware.

4) Observations / Conclusions / Comments

As I have two GD77s and I had already messed a lot the first radio, I used the other (immediately after flashing the new version) to do these tests. I realized that the problem that I reported as being of scope of the TSO, was about some other situation (I apologize!), which caused (in the lateral navigation) that all the TG would have the TS = 1 (as in my TG = 724, which has TSO = 1 - When the radio is turned on, it "wakes up" on this TG!). Only after a ClrTS did the tests above work. Perhaps it is all a matter of initializing the radio by forcing the firmware MTSet = disabled or else a question of understanding on my part (maybe, simply, I should use ClrTS more often ...).
I confess to being somewhat confused about priorities. TSO has a higher priority than ChTS, I know that. I also believe that the priority of MTSet in relation to TSO and ChTS is higher, but I am not sure if its scope should extend beyond the current TG, persisting in vertical navigation (Up, Down, Function-Up, Function -Down). In my tests, it seems that only the side navigation has the desired behavior, just navigating "outside" of the TG for MTSet to return to MTSet = disabled. Obviously, I don't know anything about the other usage scenarios and their requirements, but I think there are no major problems with my proposal.

Regards, Paulo

rogerclarkmelbourne, 2020-10-21:
Re: I press Down and the screen returns to TG = 724, but TS REMAINS TS = 2

Strange

I don't see this in the version I'm running

But I have been making some modifications, so perhaps there is some minor difference.

I agree. If the TG has a TS override e.g. your TG 724 is TS1, then when you change channel to a TG with this override then it should change to that override.

BTW.

Do you use TG Lists on your channels, or is the TG List set to "None" and you use the Contact on the Channel for the TG ?

PedroPauloBrazil, 2020-10-21:
I use only TG Lists, no contacts anymore, Roger.

rogerclarkmelbourne, 2020-10-21:
OK

Daniel and I are working on some changes.  I may post a test version here if it seems to work OK.

PedroPauloBrazil, 2020-10-21:
> OK
> 
> Daniel and I are working on some changes. I may post a test version here if it seems to work OK.

Ok, thank you!

rogerclarkmelbourne, 2020-10-21:
[OpenGD77_TS_Override.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/5418821/OpenGD77_TS_Override.zip)

Can you try this version.

It has some displays which we may not keep in the final release, but at the moment, if the Contact TS is being used, the display of TSx changes to CSx , also if there is a manual override, the TSx is displayed in **Bold font**  

I dont know if we will keep either of these displays, because I see on the forum, people do not like the bold font
( https://www.opengd77.com/viewtopic.php?f=7&t=1337#p11808  )

rogerclarkmelbourne, 2020-10-21:
@tattoli 

In the previous version, it was constantly clearing the override, which I think was a mistake, and for me it did not even set the TS for the TG.

Hence I removed the code which constantly reset the TS override.

From the forum,  https://www.opengd77.com/viewtopic.php?f=7&t=1337&p=11811#p11805   

Most people seem happy with the current implementation, but I know that @PedroPauloBrazil  found a bug, where the TG TS was not always being applied.

We will fix this bug.

PedroPauloBrazil, 2020-10-21:
Ok, Roger, I'm not in my house now, but later today I'll try the firmware. Thanks again!

rogerclarkmelbourne, 2020-10-21:
OK.

Actually I don't know if this version is perfect, I think it may now be too complicated, but it may fix the bug for you.

rogerclarkmelbourne, 2020-10-21:
Actually can you try this version.

I think its perhaps better

[OpenGD77_TS_Override_V2.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/5418946/OpenGD77_TS_Override_V2.zip)

I removed the bold font, because the feedback from the forum is that the bold font is not good

Also, when a TG has a TS setting, the internal Manual Override setting is set to this value.
So if you change to a TG which does not have a TS, then the last TS is retained.

However, I think this is where some operators will want the TG TS to be held for subsequent operations, and some people will want the Manual Override value to be cleared when the TG has a TS

Probably what is worse if that the internal Manual Override TS value (normally set by the * key), is not obvious to the operator, so retaining the internal Manual Override TS at whatever value it was by pressing * (possibly many times pressing *), will cause confusion.

Umm, I'll do 1 more version for you to try, where the internal Manual Override TS is cleared when the TG has a TS

rogerclarkmelbourne, 2020-10-21:
[OpenGD77_TS_V3_Manual_OverrideTS_cleared_by_TG_TS.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/5418975/OpenGD77_TS_V3_Manual_OverrideTS_cleared_by_TG_TS.zip)

OK.

This version my be better (depending on your operation)

In this version, when you select a TG which has a TS override, e.g. Left or Right arrow, or changing to a channel with a TG that has TS.

Then the internal Manual Override TS number is cleared. i.e so any press of * , TS value is cleared.

I think perhaps this is the easiest logic for people to understand. 

Anyway, see if you prefer V2 or V3.

Or perhaps none of them fix the problem for you ;-)

PedroPauloBrazil, 2020-10-21:
Ok, Roger, now I have a lot to "play" with here.  :) As I should get home a little later than I anticipated, I will only be able to start the tests, but only tomorrow will I complete them and report the results to you.  I add that bold does not disturb me, being a very smart way to signal that there is an active override manual.  Anyway, I will test the 3 versions and give an opinion on which one I prefer.  Best Regards!  Paulo

rogerclarkmelbourne, 2020-10-21:
@PedroPauloBrazil 

No worries

The V2 and V3 are probably the best.

The first version has the extra display of bold font, but we will probably not use bold font in the final version, as people don't like it and also I don't think it helps a lot to understand the override system.

I will probably send V2 and V3 to some other people to test

PedroPauloBrazil, 2020-10-22:
Good morning, Roger, follow the results of the firmware tests, including my impressions and suggestions. I will consider the bug resolved if the error I described in my previous message does not occur in item 3.2.4.
I observe that I have not tested other features in the firmwares, only the bug in question.

1) Version 1

Tests show that the TS override bug has been fixed. The strategy of showing CSn, TSn or TSn in bold (n can be 1 or 2) is very interesting, giving a clear view of what is happening. Personally I have no problems with bold, but if you don't want to use it, I suggest using "cSn" for Channel TS override, "tSn" for Time Slot without overrides (channel time slot, for example) and changing the TSN in bold for "mSn" for Manual Time Slot override.

2) Version 2

The bug has been fixed, but the user does not know whether the TS Manual override is active or inactive. I agree with all the comments you made in the message you sent me this version. In order to improve information for the user and for clarity, the suggestions I presented in the version 1 test remain (use cSn, tSn and mSn).

3) Version 3

The bug has been fixed, however in addition to not making it clear to the user whether a Manual Override is active, the manual override is finalized for all timeslots when navigating to a TS that has an override. Of all the versions, this was the one I liked least.

4) Comments and suggestions

I would stay with version 1 (or 2), changed to not present the bold, but to detail the current voerride mode, according to my observations in item 1.
 Another possibility that seems to me better and simpler, is to make the manual time slot override active only while in the current TG. In this case, when navigating to any TG, this TG would assume the slot as defined in the tg's 1-time slot override or channel's 2-time slot. After that, if the user wishes to manually change the slot of that channel to something different, that change would prevail, but only as long as one does not navigate to another TS, Channel or Zone. The indications on the screen could be the same as cSn, tSn and mSn suggested above. What do you think about?
 
Regards, Paulo

rogerclarkmelbourne, 2020-10-22:
Hi Paulo

Thanks for the feedback

I think the problem is that we have many different classes of users.

Some people may find the TS display too complicated if it had upper and lower case letters and also bold and not bold.

However we could possibly add an option to control this, in the Display Options, so that people who need to know this information can select that it is displayed.

BTW. 
In all 3 versions, there is a bug, where if you Manually Override the TS on a Contact (TG) using the * key, then turn the radio off and on, the TS changes to the TS of the Contact (TG).

Daniel has just sent me a fix for this, which I will test later.

Anyway, thanks again for testing, and at least you can use a version 1 until we can fix this remaining bug and also modify the display options.

PedroPauloBrazil, 2020-10-22:
Hello, Roger!

Thank you again for all the time dedicated to this subject!
Yes, I fully understand the scenario you described and once again I agree that because it is a heterogeneous environment, any changes to the interface should always be very well thought out. So, I think that your idea of ​​adding an item to the display options is excellent. Regarding the use or not of bold, "lowercase mixed with uppercase", etc., these were only suggestions. I trust that the end result will be very good, as usual.
Finally, about the bug that Daniel fixed (Thank you too, Daniel!) I didn't really notice. I'm getting old and I fear that my "career" as a software tester would be seriously compromised (if I wasn't already retired), LOL

Once again I leave you all my best regards, with the wishes of much health, peace and prosperity!

Paulo - PU2OIP

rogerclarkmelbourne, 2020-10-22:
Hi Paulo

No worries
You did a very good job of testing.

And definitely, because there are a wide variety of users, all changes to the UI need to be considered very carefully.
Not many people take this into consideration when suggesting or requesting changes or new features etc

Also because the firmware runs on the Baofeng RD-5R which has a smaller screen, this also imposes some additional limitations.

Stay safe.. 
We have been in lockdown here in Melbourne for the last 2 months +, and COVID-19 is almost , but not completely under control after our second wave..

PedroPauloBrazil, 2020-10-22:
I sincerely hope that we will all get rid of or learn to control this pandemic, my dear friend.  In the meantime, how good we have the radio and the internet to keep in touch! 73!

rogerclarkmelbourne, 2020-10-27:
The changes from Daniel are now in the pre-release version

https://github.com/rogerclarkmelbourne/OpenGD77/releases/tag/D2020.10.27

They were also in yesterday's version.

The TS display font is no changed, and it does not display CSx but the underlying functionality is the same as the test versions I posted here

closed 2020-10-27 by rogerclarkmelbourne

PedroPauloBrazil, 2020-10-27:
Thank you! I'll install and test later.

Regards,

Paulo