rogerclarkmelbourne/OpenGD77 #829: Report that not all 16 chars of a TG name is displayed when its looked up ??

rogerclarkmelbourne, 2020-10-20:
https://www.opengd77.com/viewtopic.php?f=11&t=1333

rogerclarkmelbourne, 2020-10-21:
I checked and this report is correct, Only 15 of the 16 characters are being displayed, when the contact is displayed via a lookup.

But all 16 characters are displayed when the Contact is normally displayed, and also the Contact list screen shows 16 characters

rogerclarkmelbourne, 2020-10-21:
It looks like a lot of the snprintf calls in updateLHItem are incorrect, as they specify the total buffer size as 16 when its 17

This doesn't just apply to the talkGroup it may also apply to the item->contact as this buffer is actually 20 bytes

f1rmb, 2020-10-21:
Okay, I will look at this.

rogerclarkmelbourne, 2020-10-21:
Thanks.

I did a quick test and it does look like the 

`snprintf(item->talkgroup, 16, "%s", buffer);`

should be 

`snprintf(item->talkgroup, 17, "%s", buffer);`

in updateLHItem in uiUtilities

Not sure about the item->contact but it probably should

rogerclarkmelbourne, 2020-10-26:
Fixed

closed 2020-10-26 by rogerclarkmelbourne