rogerclarkmelbourne/OpenGD77 #831: Low priority. Add Band limits option to use ranges from CPS

rogerclarkmelbourne, 2020-10-22:


rogerclarkmelbourne, 2020-10-22:
Note.

The range limits screen would also need to be enabled again in the CPS

rogerclarkmelbourne, 2020-12-22:
This was implemented several releases ago

closed 2020-12-22 by rogerclarkmelbourne

RipperFox, 2021-04-07:
Band limits from CPS (R2021.03.02.01) somehow don't work for me correctly in FW D2021.04.03.01 on my GD-77 in VFO mode: 
- Set limits in CPS to 144 - 146 & 430 - 440 MHz
- Band limits: On
- Still can TX on VFO A/B from 144 MHz up to 148 MHz and from 420 MHz up to 450 MHz

rogerclarkmelbourne, 2021-04-07:
I just tested this feature and its working fine.

Be aware if the band limits exceed the RF hardware limits they will be ignored, and the firmware will default back to 144-148 430-450

Also. The source code in this repo is not up to date.

I no longer immediatly push the latest code for the firmware to this repo, this decision was taken for a variety of reasons, including people ignoring the licensing conditions etc,  which have been discussed at length on the OpenGD77 forum.

ajorg, 2021-04-07:
> The source code in this repo is not up to date.
> I no longer immediatly push the latest code for the firmware to this
repo, this decision was taken for a variety of reasons, including people
ignoring the licensing conditions etc, which have been discussed at length
on the OpenGD77 forum.

Roger, you can't publish firmware builds without publishing the source code
for each build. The license requires you to publish the source code, and
you do not own the copyright for the full source code, so you can't pretend
to different license terms.

RipperFox, 2021-04-07:
It seems I really had not set the band limits option correctly - thanks for clarifying.

    The source code in this repo is not up to date.

Ah, but I looked at the code in the ZIP file! And just re-checked menuOptions.c OPTIONS_MENU_TX_FREQ_LIMITS:  Can only be set to "on" and off".
It's a little annoying that the github repo isn't up to date but I can understand your reasoning. 
But please: What's the point of releasing a "source ZIP file" if it's contents do not represent the state of the binaries?