rogerclarkmelbourne/OpenGD77 #832: DTMF Dialer Mod

g7jiq, 2020-10-24:
Could DTMF entry be modded to allow a string to be entered in the fly, and sent with just a PTT.

And could a in a future radio update, would it be possible to add and save to the DTMF Contact list via the radio.

rogerclarkmelbourne, 2020-10-24:
You can already press the number keys while the PTT is pressed. This transmits the tones.

The firmware memory in the radio is extremely limited its impractical to implement every function from CPS into the radio.

g7jiq, 2020-10-25:
I understand during PTT DTMF can be sent directly one number at a time, but entering 7 or 8 digits takes a bit of remembering, and if missed by the allstar node need to be ented again and again.
I just thought it would be good to be able to enter the complette string (like a phone) and then hit the PTT.

closed 2021-04-20 by rogerclarkmelbourne