rogerclarkmelbourne/OpenGD77 #834: HamLib Intergation (Runtime controling protocol)

FlyGoat, 2020-10-26:
Hi there,

I'm a new HAM waiting for callsign allocation with a little bit of low-level development skill.
I'm trying to turn my DM-1801 into a remote FM TCVR. However, after had a glance at OpenGD77 CPS code, I realized that we have no way to control radio settings at normal mode.

Also, I found another program that usually used to control radio settings, which is [HamLib](https://github.com/Hamlib/Hamlib) . RIG vendors usually have a command based protocol via DATA Link to change radio settings at runtime.

So, is it possible to create a protocol for OpenGS77 as well? I think it can be based on CDC-ACM has a shell-like command interface for users and controling software.

Thanks

rogerclarkmelbourne, 2020-10-26:
A CDC-ACM control protocol could be implemented, but the USB connection currently only supports CDC-ACM and not an audio device.

You would need to make a special cable has separate connections for the USB and the Tx and Rx audio to use this radio remotely.

Its probably possible to re-write all the USB system to be a composite USB device, but it is not easy and AFIK NXP do not provide and example of how to implement a composite device.

Also on FM, AFIK there is no hardware path for the CPU to sample the audio output from the transceiver chip.

The only hardware audio paths are to the speaker and to the C6000 to decode the 4FSK modulation.
The only semi direct path for audio is via the DTMF tone detection system, but the audio quality via this method would not be good.

AFIK, the C6000 chip does have some FM capability, so possibly there is some way to configure it make the audio samples available to the CPU, but the data sheet does not provide much information about the FM functionality of the C6000 and the FM functionality is not used by the official GD-77 firmware.

FlyGoat, 2020-10-26:
> A CDC-ACM control protocol could be implemented, but the USB connection currently only supports CDC-ACM and not an audio device.
>
> You would need to make a special cable has separate connections for [...]

Thanks for your response. My initial idea was to use Computer's Audio card to sampling analog signal. USB port is just for controling.
I'll experiment this idea in following days.
It is pity that C6000 can't stream FM audio signal via I2S, though it may be possible in the hardware block digram, I guess it is limition of C6000's internal mask firmware.

FlyGoat, 2020-10-26:
Btw is it possible to override PTT in software? I'm a little bit confused by the structure. I thought PTT button was diretctly connected to AT1846S PTT pin in FM mode, but I saw VOX was implemented in software via ADC.

rogerclarkmelbourne, 2020-10-26:
PTT is just a button connected to the MCU.

So yes. PTT is controlled via the firmware.

BTW. Perhaps you can look in the data sheets in Chinese, to see if you can tell us how to configure the C6000 for FM use?

https://github.com/rogerclarkmelbourne/OpenGD77/tree/master/docs/data_sheets/C6000

The data for the C7000 may also be the same to control the FM part of the chip, as the C7000 is a C6000 with integrated CSky MCU

FlyGoat, 2020-10-26:
> 
> 
> PTT is just a button connected to the MCU.
> 
> So yes. PTT is controlled via the firmware.
> 
> BTW. Perhaps you can look in the data sheets in Chinese, to see if you can tell us how to configure the C6000 for FM use?

The datasheet told us how to use "FM Analog Channel in appendix A", which I guess is the FM XCVR part.

```
- Config registers: 0x06[1]＝0，0x10[7]＝1 to trun off the ringtone playing channel and enable FM mode.

- Set 0x06[2] to select external or internal codec.

- After selecting the codec, you have to further config codec subsystem. Please see ringtone channel's document for deatils.

- Write 0x37＝0x00 to ensure DACDATA is unchanged.
```

And for the paragraph describing the ringtone channel:
```
- ringtone mainly including poweron poweroff message etc

- Set 0x06[2], to select external or internal codec.

- If you're using external Codec，you'll have to set 0x36[7]＝0，to enable Codec I2S BCLK LRCK pin；config 0x30、0x31 for BCLK freq，config 0x32、0x33 for LRCK freq，See Codec  I2S RW timing for details.

- For internal Codec, write 0x0D＝0x10，Inernal Codec will control AD/DA switch in normal mode，when internal Codec is not in-use this register can be used to set it into low-power mode；write 0x0E＝0x8E，to enable MicEn、HPoutEn、LineoutEn，HPoutVol should be 0dB defaultly，please adjust it by your needs；write 0x0F＝0xB8，default ADLinVol and MicVol is 0dB,配write 0x38＝0x00, don't modify it.

- Write 0x37＝0x00 to ensure DACDATA is unchanged.

- Write 0x06[1]＝1 to ture on ringtone.

- Then you must write data by U_SPI to Codec immediately to play audio.

- After the audio ended write 0x06[1]＝0 to turn off ringtone.
```

For CTCSS and DTMF part, I'm still trying to understand them. They were written in a really bad manner. Although I've dealt with many crappy Chinese datasheet, this one is extremely hard to understood. Perhaps because they assumed readers should have deep understanding on radio.... 

> 
> https://github.com/rogerclarkmelbourne/OpenGD77/tree/master/docs/data_sheets/C6000
> 
> The data for the C7000 may also be the same to control the FM part of the chip, as the C7000 is a C6000 with integrated CSky MCU
> 
> https://github.com/rogerclarkmelbourne/OpenGD77/tree/master/docs/data_sheets/C7000
> 
> I am also interested if its possible to access the direct 4FSK decoded signal in the C6000 / C7000, so if you can find 
information on how to access this data, it would be very interesting, as it may be possible to software decode other modes, e.g. P25 and NXDN etc

Datasheet didn't mention anything about raw 4FSK.  :-( But for C7000 I'll try to ask C-SKY if they know further details. I knew some guys work at C-SKY.

FlyGoat, 2020-10-26:
I assume the ringtone channel is what OpenGD77 used at `functions/sound.c`.

rogerclarkmelbourne, 2020-10-26:
@FlyGoat 

AFIK
Sound.c does not use ringtones, it sends raw audio via I2S to the C6000

Perhaps they call this "ringtone"  I don't know.

rogerclarkmelbourne, 2020-10-26:
Re: CSKY

I don't think they will know, because CSKY is just the MCU which is embedded inside the C7000.

Probably the internal C6000 interface / registers did not change when they added embedded the CSKY MCU.

rogerclarkmelbourne, 2020-10-26:
BTW.

4FSK may be called something like Physical layer ?

FlyGoat, 2020-10-26:
> 
> 
> @FlyGoat
> 
> AFIK
> Sound.c does not use ringtones, it sends raw audio via I2S to the C6000
> 
> Perhaps they call this "ringtone" I don't know.

I checked the register written here, it's called "ringtone mode" in the manual.
So maybe we can play audio on FM in this way :-)

rogerclarkmelbourne, 2020-10-26:
Tx modulation is handled in a strange way.

the C6000 controls the master clock oscillator to make the 4FSK data, rather than sending that audio signal to the AT1846S mic input

FlyGoat, 2020-10-26:
> 
> 
> BTW.
> 
> 4FSK may be called something like Physical layer ?

In the manual, it said `User will be able to retreive physical layer frame data in single layer mode`, but didn't mention `single layer mode` elsewhere.

I'll try to dig it deeper!

FlyGoat, 2020-10-26:
The single layer mode is mentioned in `Bit error rate experiment section`.
It said.

```

This experiment will let HR_C6000 continuously to receive 4FSK signal，the freq is configurable，455 kHz or 450kHz is recommended. HR_C6000 will store decoded 36 byte data frame at a region of 1.2KRAM start from addr 0x30. MCU will be able to read the frame from RAM via SPI port and compare it with the sended frame to calculate HC_C6000 error bit rateperformance.  The structure of data fame stored in RAM is shown in 5.15，MCU should determine frame type by interrupt [flygoat:???? there is only one "physical layer interrupt in the previous part of manual."] (SyncState and SyncClass(0x51) and decode them according to the lengh [flygaot:?????] (dotted line parts in the figure needs to decode extra CACH data) . 
```
With a register table at Page 51.

I had a impression of how to receive physical frame but sending part is still not clear.

rogerclarkmelbourne, 2020-10-26:
OK.

I tried to read a google translate of the data sheet, but the only references to reg 0x51 seem to be to read the DMR data frame type.

I also read that there are some "test" modes, but it was unclear how to use these.

We thought the problem with the data sheet was the automatic translation from Chinese to English, however I think you are saying that the data sheet is not very useful even for people who can read it in Chinese.

I think there must be other data sheets for the C6000, however this is the only data sheet we were able to obtain.

Originally the C6000 was manufacturer by Hongrui , and they did not publish any data sheets.

I contacted Radioddity and they tried to contact Hongrui  to obtain better data sheets, but this was not possible.

I don't even know if TYT has the full data sheets, because its possible that the firmware for the GD-77 / MD-760 (and Baofeng DM-1801 etc) was all written by programmers who worked for Hongrui .

I have examined the official DM-1801 firmware in detail and the original source code base for it is definitely the same base source code used by TYT in the MD-760

Also, the bootloaders in the MD-760 and Baofeng DM-5R and DM-1801 is almost identical.

So the programmers who wrote the firmware for all these radios probably also wrote the bootloader.

tangogc, 2020-10-29:
wonderfull I vote for this integration

FlyGoat, 2020-10-30:
I'm still working on it. I will prepare a pu request for comments when it's ready.

rogerclarkmelbourne, 2020-10-30:
No rush.

I've very busy at the moment, integrating other features.

Also, as code space is very limited in the radio, I may not be able to accept the submission of there are other higher priority functions which need the program space.

Specifically SMS functionality is currently under investigation by a developer in Germany, and he may send me his code when it is fully working.

FlyGoat, 2020-11-01:
Any idea about DM-1801's SWD pinout?
My DM-5R was too crappy to have USB connection and TX at the same time so I just got a DM-1801 but failed to locate it''s SWD pins to get J-Link work :-(

rogerclarkmelbourne, 2020-11-01:
I thought the SWD pins are labelled on the PCB. If not you will need to track the connections to the MK22.

Also. You need to do a complete erase of MKU to de-protect it.

Make sure you backup the MCU ROM using the CPS.

To erase the MK22 you need to connect the Reset pin as well as SWDIO and SWCLK.

After the MK22 is totally erased, you can program the backup you saved using the CPS, and it will then be fully accessible.

FlyGoat, 2020-11-26:
Spending many times on that I found there are some issues on Mic.
When USB is connected my DM1801 will just TX random noise instead of voice from mic of Kenwood connector.

Any idea?

Thanks.

rogerclarkmelbourne, 2020-11-26:
@FlyGoat 

I don't know. I don't have a cable that has USB and audio connections.
And I don't have a schematic for the DM1801 :-(

closed 2021-04-20 by rogerclarkmelbourne