rogerclarkmelbourne/OpenGD77 #835: Bug: when voice prompt is set to 3, the voice reads when the PTT is depressed.

marrie, 2020-10-27:
# steps
1. Press menu, up arrow 4 times to sound, press menu, up arrow once to voice prompt.
2. Choose prompt 3.
3. Confirm with menu, and keep confirming until you are in the channel again.
4. Press the ptt key and let go.
5. Notice the voice reads.

This should not be the case. I'm using 161426oct262020 b2be69b. I think I copied that correctly. I don't have a work around at the moment, but I hope this can be taken a look at. I could not reproduce that in the build from the 25th of October.

Thanks.

vk7js, 2020-10-27:
Thanks Sarah.

Joe

rogerclarkmelbourne, 2020-10-27:
Already reported by Ian DJ0HF yesterday and already fixed in the latest development release

https://github.com/rogerclarkmelbourne/OpenGD77/releases/tag/D2020.10.27

closed 2020-10-27 by rogerclarkmelbourne