rogerclarkmelbourne/OpenGD77 #840: CPS channel import is not possible

DC6PA, 2020-10-29:
In CPS it is possible to export channel information, but it is not possible to import this file back to CPS.
I used this function before, to add new channels or to change the order of the channel items.
It would be nice to change the order(position) of in the channel, in the channel table!

f1rmb, 2020-10-29:
[http://www.opengd77.com/viewtopic.php?f=7&t=1346&sid=18ce9074a54840010e42e4b1207b7371](url)

DC6PA, 2020-11-03:
Channels
=================================
Name | Ch Mode | Contact | TG List | Remark
Channel_1 | Analog | N/A | None | ok
Channel_2 | Digital | TG 9 | None | ok
Channel_3 | Digital | TG 9 | TG-all | ok
Channel_4 | Digital | TG 9 | None | ok
Channel_5 | Digital | N/A | TG-all | error

if you klick on the row header to edit the item in the detail screen, the value of the Contact of this row will be reset to "N/A".
If you export this values now and try to import it again, you get an error message if TG List is set to not equal "None".

rogerclarkmelbourne, 2020-11-24:
Use.
http://www.gb3gf.co.uk/downloads.html

closed 2020-11-24 by rogerclarkmelbourne