rogerclarkmelbourne/OpenGD77 #846: Bug: ptt latch does not appear to work when the option is switched on

marrie, 2020-11-02:
# summary
When going to menu, options, ptt latch and letting it to on, then hitting menu, and exiting the menus via the cancel button, ptt latch does not work.

# steps
1. Press menu.
2. Press up or down until you get to options then press menu.
3. Press up or down until you get to ptt latch off.
4. Press right to turn it on.
6. Press menu to confirm your changes.

when you press the ptt key once you get out of the menu and back into your channel, and you unkey,, you will hear your link tone, or if on a repeater, its tone.

am I missing something here? I'm  running 140054nov120202 16a791a

rogerclarkmelbourne, 2020-11-02:
Please can you explain what you mean 

"when you press the ptt key once you get out of the menu and back into your channel, and you unkey,, you will hear your link tone, or if on a repeater, its tone."

Steps to contact a repeater with PTT latch on

Press and release PTT
Radio enters Tx and stays in Tx
Radio attempts to contact repeater
When radio connects to repeater , Tx start beep is played
If radio is unable to contact repeater, it keeps retrying indefinitely (this is not ideal and should have a timeout)

When radio is connected to repeater, press PTT again to release the latch.

rogerclarkmelbourne, 2020-11-02:
BTW. What Prompt level are you on.  Looking at your previous posts perhaps you have Voice Level 3 enabled.

marrie, 2020-11-02:
That's the problem. I turn on ptt latch, but when I get out of the menu, 
I when I tap the ptt key it will let go right afterwords. I know this as 
I hear the default link tone I have set up. [Here is a dropbox  link of 
a test. 
](https://www.dropbox.com/s/u9y85siszlv436p/speaker%20mic%20almost%20working.mp3?dl=1)Skip 
past the first few minutes where I am just testing the speaker mic to 
see if it actually works. You will hear the key let go and not stick. I 
was actually connected to my node at the time, not to anything  on the 
node but just the node part.

marrie, 2020-11-02:
I'm actually on  prompt 1.

Thanks.

rogerclarkmelbourne, 2020-11-02:
Do you have 2 radios and can test in FM.

Single press should start Tx, press again should stop Tx.

I don't have a simplex hotspot, but I have not received any reports of problems about this function, so you'd imagine it is getting used by other people.

We have had problems, not on PTT latch  - but its the same thing, of external speaker mics with loose connections, doing muliple triggers to the PTT, which would key it on and then immediately off.

However, the internal PTT doesn't seem to have this problem,.

marrie, 2020-11-02:
I don't have a second radio but when I listen through  IAXRPT  i also 
hear it unkey. I'll check again tomorrow before I hop off to work. In 
theory what is supposed to happen?

Thanks.

rogerclarkmelbourne, 2020-11-02:
Yes.

Press PTT to start Tx
Press PTT again to stop Tx

marrie, 2020-11-02:
Copy that. I did a quick test and [Here are the 
results](https://www.dropbox.com/s/src7n8bipr07dm7/ptt%20latch%20fail.mp4?dl=1).

I turned  on ptt latch from the menus, then pressed the ptt once to 
start, but heard the  tone, not a continuous transmition. I get how this 
is supposed to work and was hoping to test this out during some very 
long transmitions I make when I run nets etc, as my hand gets tired 
easily.

I did not convert the file example  to mp3, and I am dead to the world 
right now, but I hope this is clear enough. The click you will hear is 
when I mute my computer mic so you can hear what's going on with out my 
noise distracting from the issue

Thanks.

f1rmb, 2020-11-02:
Hi,

   PTT Latch depends of the TOT of the channel. It no TOT value is set, the PTT Latch won't work (check the user manual, it's a safety measure).

Cheers.
...
Daniel

rogerclarkmelbourne, 2020-11-02:
Hi Daniel

Thanks. I'd forgotten that.

rogerclarkmelbourne, 2020-11-02:
BTW. There is still a small bug, the TOT doesn't start counting, until the radio contacts the repeater.

So if it can't contact the repeater, it keeps retrying indefinitely

f1rmb, 2020-11-02:
Hi Roger,

   Okay. How long it should retry ?

Cheers.
...
Daniel

f1rmb, 2020-11-02:
Okay.

rogerclarkmelbourne, 2020-11-02:
I think the normal value is 3 retries.  (so 4 pulses in total)

marrie, 2020-11-02:
Ok.

Good catch. Unfortunately my transmit overtime actually turns off when I shut off the radio. I have it set to about 45 seconds. But it resets to off when I turn off the radio and I have to set it every single time. I don't mind doing this, but I keep forgetting to do this. Is this a separate bug? I will try with a transmit overtime of 15 seconds today just to be sure. Just connected to my all-star node, not my All-Star note connected to anyone else just in case I interrupt traffic. Which I really don't want to do. It does show as 45 in the software on my computer. It is not written to memory, that's fine though. 

So, assuming I can get this working, will there be a sound in my earphones when I'm monitoring on I a XRPT that I am actually going? Or will I hear the carrier wave when I start?￼￼￼

f1rmb, 2020-11-02:
Hi,

   I think you really need to read the manual, there is a PDF that you can probably use with some specific screen reader device:
[Link to PDF version of the User Guide](https://github.com/rogerclarkmelbourne/OpenGD77/blob/development/docs/OpenGD77_User_Guide.pdf)

if you change any Channel detail, and you want to keep it (after a power cycle or channel change), you have to save it back to the codeplug by pressing SK2 + GREEN. Pressing GREEN only is set a temporary change.

Cheers.
...
Daniel

marrie, 2020-11-02:
that's fine for now.

with my TOT set at 15 seconds, the ptt latch still fails. [sort of the same example as last night](https://www.dropbox.com/s/npkyq6s5qvydfrk/ptt%20latch%20fail%2002.mp4?dl=1)

Again I'm connected to my allstar node only, and not any other node from my allstar node as I don't want to flood them with random key ups. This should work regardless, correct?

marrie, 2020-11-02:
Ok, I got it to work with a TOT of 15 seconds. That's fine actually for me to set it every time, this way if I don't have it set, I can be as normal. [example of the working ptt latch is here,](https://www.dropbox.com/s/swimodf9z3u0hhc/ptt%20latch%20success%20almost.mp4?dl=1) although I heard a bit of a click when I pressed the ptt button again. I wish there were a bit of a beap that would be audable only to me that would not go over the air, but this will work for now.

good catch on the TOT setting, and I am in voice mode prompts 1. Suggestions on how we can make ptt latch better for those who cannot see, or cannot read the screen for any other reasons? as we might not know when we are transmitting as we cannot see the "connected" message if one does show up on screen when we tap the ptt button.
Thanks.

rogerclarkmelbourne, 2020-11-02:
Its not practical to make the firmware be able to respond to all miss configurations that the operator may use, with detailed feedback about the nature of the error.

In this case , we are now adding a general purpose error display which plays the error beep sound, which is used for other errors and hence we may able to trigger it under the PTT latch with No TOT set. Just to report "ERROR"

The problem is the severely limited program space in the radio, with only a total of 200k bytes available for all functionality and texts.
Also we are starting to run into memory limits with the voice prompts, stored in the Flash memory chip, as there was only around 128k free in that chip and we now use around 95% of that capacity, so adding many extra voice prompts would require reductions in the other things which use memory like the DMR ID database

marrie, 2020-11-02:
I don't know if my email reply made it so am trying again.

the beep is fine, at least it will serve as an audible indication. 200k of memory is not a lot to work with. I'm just glad this works.

Thanks so much.

closed 2020-11-02 by marrie

rogerclarkmelbourne, 2020-11-02:
BTW.

I'm not sure why you have a problem setting the TOT, This can be set in the Channel Details, and permanently saved back to the codeplug by using SK2 + GREEN, as documented in the User Guide.

TOT values can be very large. I think the max value is 3825 seconds, which is well over an hour, and I think the battery is likely to go flat before the TOT times out.

However, we may need to modify the PTT / TOT error handling, and possibly not permit latching if the TOT is set to a very high value