rogerclarkmelbourne/OpenGD77 #847: No CTCSS decode

g7jiq, 2020-11-03:
Just updated to the latest D2020.11.03 and I have found that the CTCSS decode on FM (VFO or Memory) no longer works, (no rx audio, tx ctcss ok)
Going back to  D2020.11.01 works fine.

rogerclarkmelbourne, 2020-11-03:
The initial upload had that bug, so I uploaded again

Try downloading again and reconfirm the bug still exists

rogerclarkmelbourne, 2020-11-03:
I just re-tested this and I don't see any problems

What radio are you using, and when did you download.

The version with the error was updated on github around 12 hours ago, so I can't see how you could have just updated in the last hour and have the version with that problem

g7jiq, 2020-11-03:
I had downloaded and programmed my radio earlyer in the day (more than 12 hours ago)
I have just re flashed and used the download list from your opengd77cps, and all is working ok.

Just to let you know I have two GD-77 radios I try the software on,

rogerclarkmelbourne, 2020-11-03:
OK.

I think, I'll need to change my procedure, so that if I upload something with a problem, and notice shortly afterwards, I'll need to immediately do to fix it and also do another release, but I'll need to change my release numbering system to allow multiple releases on the same day,  e.g D2020.11.03.01   but goodness know how this will be handled by Github, as it does not like the numbering system being changed

g7jiq, 2020-11-03:
Yes I agree with you, 
Muliple releases would allow someone to swap and change, to help try and narrow down a bug or a feature.

I don't think you would have any problems with Github, as long as you change or increment the ver number.
I follow the UHSDR Project Page (mcHF) on Github and in the past there has been multiple updates with no problems.

rogerclarkmelbourne, 2020-11-03:
I'll close this now as the bug no longer exists

closed 2020-11-03 by rogerclarkmelbourne