rogerclarkmelbourne/OpenGD77 #849: Radio will not go out of squelch or with too low receive signal until you press PTT  - GD77

SP4SWD, 2020-11-04:
Radio: GD77 with openGD77 firmware dev 03.11.2020 also from june had the same issue.
Analog mode.
Squelch set at ~20%
I am measuring power with builtin RSSI(not precise but gives clue what happens)
How the issue looks like:
Lets have a 2m band freq as 145.4 Mhz listening no difference VFOA or B
i got noise level at -110 dbm - squelched.
Then I present weak signal at -103dbm and the squelch does not open - the radio DOES NOT show that there is any signal the measured RSSI does not go up.
And now is the magic trick - press PTT for very short period(just touch less than 0.5s) and release - SIGNAL is recognized by GD77!? The same present this signal - it is on channel - checked by other radio and change frequency and go back - VIOLA signal is showing up. 

Second sittuation - 
The same noise level (-110dbm) - no signal. Radio is squelched 
Signal is beeing rised by 10db on transmitter.
Signal is beeing transmited. Ref radio sees the signal.
GD77 opens squelch also and the signal shows up. Nice but... radio is showing low signal of ~-104dbm. Then magic - quick press PTT and release it -signal after releasing jumps about ~10db to -92db!!! Changing frequency when radio is on this channel and going back fixes reception.

This is realy an issue when i am listening to certain channel. 

What is the problem ? What is the solution ? Radio is working normally. C

SP4SWD, 2020-11-05:
This is realy an issue when i am listening to certain channel. - i meant while i am not in scanning mode. There it works until squelch does not engage.

rogerclarkmelbourne, 2020-11-05:
Is Rx CTCSS enabled ?

rogerclarkmelbourne, 2020-11-05:
[OpenGD77_test_2020110501.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/5491729/OpenGD77_test_2020110501.zip)

Can you also install this version, so that I know you are using the same version that I am currently using.

I can't replicate your problem either with CTCSS on or off.

You also didn't state what mode you were using,  my guess is you are using FM, but I can't be sure.

SP4SWD, 2020-11-05:
Same issue with firmware you have provided. TX CTSS off, RX CTSS off, FM mode, all CTSS/DCS off, filtering CTSS/DCS off or on - does not change behaviour. Simple FM simplex. Radio is still not sensitive until you change frequency or you press and release PTT. Problem is with weaker signals. Watch the video: https://www.youtube.com/watch?v=cWbYx0D7a3M signal on opengd77 first -104dbm after pressing PTT -96dbm and stable... It looks that baofeng UV5R is better than gd77 at plain FM signal reception ? Signal is generated with hackRF one. I was thinking - GD77 saturates... But why after pressing PTT it does not saturate? When you change frequency it does not ? When I connect UV5R to antenna - NO such issue. No strong stations nearby. Will try with stock firmware on GD77. I spotted this issue while talking in group with weak station and i spotted that I can hear weak station after I press and release PTT, also spotted that other stronger stations are not so strong as after I release PTT and squelch does not engage. Very strange issue.

f1rmb, 2020-11-05:
Can you zip and post your codeplug ?

Cheers.
...
Daniel

SP4SWD, 2020-11-05:
Look at codeplug.
[SP4SWD_0731_175937.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/5492941/SP4SWD_0731_175937.zip)

SP4SWD, 2020-11-05:
Just checked with stock firmware of GD77(cant measure RSSI unfortunately - or tell me how ?) at same conditions and squelch at GD77 is released now first, signal and squelch releases always other than at UV5R which can not unlock squelch sometimes at the same conditions... Sensitivity tests so far: stock GD77>UV5R>OpenGD77(not until freq change or PTT press/release) ?

SP4SWD, 2020-11-05:
Oh and stock firmware is 4.3.3

rogerclarkmelbourne, 2020-11-05:
Stock software can't measure RSSI in the GD-77.

f1rmb, 2020-11-05:
I've made few tests, using SP4SWD codeplug, etc, and I'm unable to replicate.

rogerclarkmelbourne, 2020-11-05:
Did you set a frequency in the VFO ?

What was the frequency ?

After you press PTT and the signal increases,  what occurs to make the signal go down again?? Change frequency, Reboot radio ??

SP4SWD, 2020-11-05:
The signal goes down when any signal dissapear. When signal shows up again it is weaker until I press PTT.

SP4SWD, 2020-11-05:
If signal is present it does not go down until signal dissapear.

SP4SWD, 2020-11-05:
frequency set in VFO A or B or channel 145.400 - no difference the same behavior.

rogerclarkmelbourne, 2020-11-05:
What is the signal you are receiving ?

rogerclarkmelbourne, 2020-11-05:
Does it happen on any band and any frequency ?

SP4SWD, 2020-11-05:
I can replicate the issue with any signal(analog) - i spotted the issue and i can simulate with transmission using HackRF one. When i spotted the issue i investigated using HackRFOne.

SP4SWD, 2020-11-05:
Band of interest is at 2m band. Will try at 70cm FM.

rogerclarkmelbourne, 2020-11-05:
Does the problem occur at 145.500Mhz

rogerclarkmelbourne, 2020-11-05:
If you transmit on your other radio on 145.400 does the signal level increase on your GD-77 ?

But I think you only have 1 radio attached to an external antenna.

You also didn't explain what the is the signal you are listening to

rogerclarkmelbourne, 2020-11-05:
Also

Got to Options menu

Select "User power:"
Press Left arrow until the value is 0
Press Green button to save

On VFO screen set power to 5W.
Then press and Hold SK2 (Blue), and Press and Hold Right Arrow

Observe power setting is now "+W-"

When there is the signal, try transmitting now, and see if there is any change to the signal level.

SP4SWD, 2020-11-05:
I returned to openGD77 firmware from stock.
The same issue is with 145.500MHz. 
Yes I got 1 external antenna 10m above my station.

Yet what I observed - When I present to the air using hackrf a weak signal, opengd77 shows signal with -100dbm(noisy) level - then I press ptt on uv5r(second radio with handheld antenna, and low power level setting, frequency the same 145.5 or adjacent channel 145.0125) - got for short period received signal of -30dbm - -50dbm at the gd77 and after releasing ptt on uv5r the gd77 suddenly receives at -93dbm(normal - lower noise level).
Another clue - strong signals also override something and make reception of weak signals a lot better after loss of this strong signal(like other radio carrier) - the same as changing frequency or pressing ptt on GD77.
https://www.youtube.com/watch?v=e2HAz84Pr-Q

The same happens with "+W-" mode. 

Still when I release the ptt/change frequency/force stronger signal at the same time. Test signal(nfm modulation) is stronger at reception. Going to lack of signal at channel makes problem. When I turned off squelch(by lowering to minimum - hearing noise at -112dbm) and presented the weak signal from hackrf i got -100dbm then I presented other strong signal simultaneously for short period and removed it - i got -93dbm of weak signal. 

It looks like problem with tuning demodulator to strongest carrier?

The problem does not occur at 70cm band - received signal strength is the same no matter what I do with ptt or frequency changes.

rogerclarkmelbourne, 2020-11-05:
If this problem happens in the stock firmware, which is completely different from the OpenGD77 firmware, this must be a hardware feature of the radio, or something caused by your testing.

The OpenGD77 firmware is not like the MD380 tools, its totally clean and does not use any code from the stock firmware apart from the DMR codec (which is only used because of license / legal reasons)

Hence if this is a function of the hardware in the radio, and also present in the stock firmware, then its not something that can be resolved in firmware.

Also, neither Daniel or I can replicate this problem, so it could be a hardware fault on your radio.

SP4SWD, 2020-11-05:
The stock firmware works normal. The same as UV5R. Problem is on opengd77. Noise level is good when i present the signal from the start.

rogerclarkmelbourne, 2020-11-05:
But its not possible to see the RSSI on the stock firmware.

Anyway. I can't replicate this issue, hence its impossible for me to fix this.

closed 2020-11-05 by rogerclarkmelbourne

rogerclarkmelbourne, 2020-11-05:
BTW.

Dont use the version I posted here, it has some other serious problems

Use the 2020.11.03 release version

rogerclarkmelbourne, 2020-11-09:
I have discussed the symptoms you described with some RF engineers, and what you describe and show in your video, is a common antenna fault.

The antenna has a corroded connection, which is acting as a varicap diode.

When you transmit the voltage across the varicap is charged by the transmission, this increases the capacitance of the varicap and it passes more signal.

Then the varicap slowly discharges, and the signal level will decrease.

Transmitting again, even a small amount of power, or connecting a DC voltage to the antenna, will have the same effect.

This happens on HF as well as VHF antenna.

The effect potentially only happens on VHF, because the varicap that was formed can't rectify UHF frequencies, i.e parasitic diodes are not likely to be able to rectify higher frequencies