rogerclarkmelbourne/OpenGD77 #852: Translating to Russian

da3020, 2020-11-08:
Hi everyone!
How can I help with the translation to Russian language?

Evg33, 2020-11-08:
https://github.com/rogerclarkmelbourne/OpenGD77/tree/master/firmware/include/user_interface/languages/

rogerclarkmelbourne, 2020-11-08:
Its not simple,

This has been discuessed before. The font does not include the Russian character set, and as the maximum number of characters in the font is 256, as its an 8 bit font, its possible that it there is simply not enough space

Also. There is a noticable lack of support for ongoing translation from the original translators of many of the languages currently supported, and in the future, if people can't be bothered to maintain the language they have added, it will be removed; as its not practical for non native speakers to update the language files.

f1rmb, 2020-11-08:
Also, if someone is willing to modify the fonts to have Cyrilic glyphs support (I can provide all the files, but there are 5 fonts), plus translation file (simple C header), it could only be supported by a specific build version, Russian only.

da3020, 2020-11-08:
> I can provide all the files, but there are 5 fonts

So we need to make correct Cyrillic character set files first, as I understand. let me try!

rogerclarkmelbourne, 2020-11-08:
I don't want to have to build a separate version for Russian.

The build process is already very time consuming.

If someone else wants to build their own Russian version, thats fine, but I do not have time to support another complete set of builds.

rogerclarkmelbourne, 2020-11-08:
Perhaps the Russian guy who did the port to the RD5R can handle all the Russian versions

da3020, 2020-11-15:
...but are any same problems with CPS translation?

rogerclarkmelbourne, 2020-11-15:
No.

CPS is on the PC and has access to whatever fonts are on the PC. The texts are stored in a XML file, which is installed in the exe installation folder, so you can just copy the XML file and easily make a new language.

On the PC you have effectively unlimited storage, unlimited fonts, many Gb of RAM etc.

In the radio, it has about 10k of RAM free and now only about 32k or perhaps 64k of program space remaining.

The processor in the radio is tiny, its much less powerful even than a PC from the 1970's

rogerclarkmelbourne, 2020-12-28:
Looks like no one picked up the ball on this one, so I'm going to close it now.

Please re-open when you have made some progress with the port to Russian

closed 2020-12-28 by rogerclarkmelbourne