rogerclarkmelbourne/OpenGD77 #853: when toggling between vfo and channel mode on DM1801, no voice prompt feedback to tell you which mode is active.

vk7js, 2020-11-10:
Please make the button either announce channel name or vfo a/b when pressed at voice prompt level 3 so the user knows which mode they are toggling to. I ended up only figuring this out by accident with lots of frustration when I received a new dm1801 and loaded on the Open GD77 firmware. After using a gd77, I couldn't figure out which button toggled between channel and vfo mode. I expected it to be the top right button as on the gd77 but it was the lower left button. For consistency between the radios it would be nice if it was the top right but if that can't be changed, please add voice prompt at level 3 to make it easier to discover.
Thanks
Joe

vk7js, 2020-11-10:
Just to clarify, please make the button which toggles between vfo and channel mode on the DM1801 provide the same feedback as the red button on the gd77.

f1rmb, 2021-01-24:
This one has been fixed.

closed 2021-01-24 by rogerclarkmelbourne