rogerclarkmelbourne/OpenGD77 #854: a Question re sk2+star

vk7js, 2020-11-10:
@rogerclarkmelbourne, When you fixed sk2+star to announce the mode, you mentioned that you could not make this symetrical, i.e. anounce fm or dmr for a reason. When I toggle this in vfo mode, I hear "mode fm" or a beep. In Channel mode, I hear "mode fm" or just the channel name. I do not hear any tg number which is what I recall you saying should be announced. Could you please clarify why this could not be made symetrical or fix it so it is? Again, just mentioning the channel name does not indicate if the channel is dmr or fm and since if you arrow away from and back to the channel you actually have no way of knowing unless you go into the details screen, this could be a confusion. Sorry to be pedantic about this but not having symetrical announcements can confuse people or at least leave one in an ambiguous state of understanding.
Joe

f1rmb, 2021-01-24:
Hi Roger,

  I think this one can be closed.

Cheers.
...
Daniel

vk7js, 2021-01-26:
This was kind of fixed. If you hit sk2+star, radio in voice prompt level 3 says "dmr" or "mode fm". Please get rid of the word "mode" so that the prompt is symetrical, i.e. DMR or FM. No need to say mode for FM, it doesn't say it for DMR.

vk7js, 2021-03-07:
THis appears to be fixed in 03/06/2021 build.

closed 2021-03-07 by vk7js